davidferguson/pibakery #1: Support for window resizing

AberDerBart, 2016-09-07:
It would be nice to be able to resize the PiBakery window to have more space for editing (tested on Windows).

davidferguson, 2016-09-07:
Blockly makes window resizing a little trickier than I'd like, but I've just had a play, and it should be possible. I'll try and add it to the next update - follow [@PiBakery on twitter](http://twitter.com/PiBakery) to get news of when I release that.

AberDerBart, 2016-09-07:
Nice, thank you very much. PiBakery is a great tool, I have been
searching for something like this for a long time.

Earl-Brown, 2016-09-07:
I just created a pull request that will at least allow the window to resize.

When resizing, the scroll bars "jump" a bit, and sometimes don't align to the window edge correctly.

Also, it would be nice to save/restore window metrics (position, size)

davidferguson, 2016-09-14:
PiBakery source now supports Window resizing; currently it doesn't save window metrics but I'll look into adding that in the future.

commit a9e71d1c (2016-09-14): window is now resizable, fixes #1

The window can now be resized. Currently it does not remember the
resize position, but this could be added in the future.
closed 2016-09-14 by davidferguson