davidferguson/pibakery #2: Linux support ?

unixfox, 2016-09-07:
Does PiBakery will support Linux ? Lot's of raspberry pi users are Linux desktop users.
I like this software and I'd like to test it on my Linux OS !

MattMattV, 2016-09-07:
It support it thanks to electron (;

MajorToMToM, 2016-09-07:
You can follow the instruction on this page at the bottom. (https://github.com/davidferguson/pibakery/). You may need to install "npm" for example by using "dnf install npm".

davidferguson, 2016-09-07:
Not everything works under linux just yet - there are a few windows and mac specific commands in these still that I'm moving over to Linux (mainly to do with detecting the SD card).

dacresni, 2016-09-08:
@davidferguson  could you point to a feature branch so others can work on that feature in particular?

Lewiscowles1986, 2016-09-08:
Pretty sure the elevate.js alone prevents Linux OS clients from working without sudo as they do not identify as darwin. Surprised to see it is not Linux OS by default with Windows and Mac as optional extra's

davidferguson, 2016-09-14:
The source of PiBakery now supports Linux - and I'll release a .deb package as soon as I can.

closed 2016-09-14 by davidferguson

unixfox, 2016-09-14:
I'll make a PKGBuild for ArchLinux when I'll have some times.

Lewiscowles1986, 2016-09-15:
> The source of PiBakery now supports Linux

I guess I should stop my similar work...

crow1170, 2016-09-18:
I've compiled from source (334ff1d) but the 'write' button appears to do nothing. Should I open this as a new issue?

davidferguson, 2016-09-18:
As it works on my linux laptop, a new issue would probably be better.
You can use the shortcut `ctrl-shift-i` to open the Developer Inspector, and if you select the `Console` tab, it might show you an error in there.

Lewiscowles1986, 2016-09-18:
I Know this is closed but I can't see another issue referencing this one. @crow1170 it looks like it's very tool-chain dependent. Probably best to run from terminal, and run through the code installing any utilities you don't have. I had a problem running as sudo due to using nvm rather than distro-installed node. Running from terminal made it easier for me to diagnose issues.

alfieoriginal, 2017-01-12:
Any news about a .deb package?

dauuricus, 2018-01-22:
I'm in trouble at `npm start`

when I try to start Pibakery on my Ubuntu Terminal...

here is Terminal 
```
blackcat@blackcat:~/pibakery$ npm start

> PiBakery@0.3.7 start /home/blackcat/pibakery
> electron .

sh: 1: electron: not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! PiBakery@0.3.7 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the PiBakery@0.3.7 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/blackcat/.npm/_logs/2018-01-22T08_32_48_213Z-debug.log
```

and this is 2018-01-22T08_32_48_213Z-debug.log
```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.4.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle PiBakery@0.3.7~prestart: PiBakery@0.3.7
6 info lifecycle PiBakery@0.3.7~start: PiBakery@0.3.7
7 verbose lifecycle PiBakery@0.3.7~start: unsafe-perm in lifecycle true
8 verbose lifecycle PiBakery@0.3.7~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/blackcat/pibakery/node_modules/.bin:/home/blackcat/.cargo/bin:/home/blackcat/bin:/home/blackcat/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle PiBakery@0.3.7~start: CWD: /home/blackcat/pibakery
10 silly lifecycle PiBakery@0.3.7~start: Args: [ '-c', 'electron .' ]
11 info lifecycle PiBakery@0.3.7~start: Failed to exec start script
12 verbose stack Error: PiBakery@0.3.7 start: `electron .`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:160:13)
12 verbose stack     at maybeClose (internal/child_process.js:943:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
13 verbose pkgid PiBakery@0.3.7
14 verbose cwd /home/blackcat/pibakery
15 verbose Linux 4.13.0-26-lowlatency
16 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
17 verbose node v9.4.0
18 verbose npm  v5.6.0
19 error file sh
20 error code ELIFECYCLE
21 error errno ENOENT
22 error syscall spawn
23 error PiBakery@0.3.7 start: `electron .`
```
in this case 19-23 are error massage.
what I do for electron finding?

dauuricus, 2018-01-22:
and my image.json file is

`[
	{
		"version": 6.0,
		"compressedMD5": "cb29576520ed28fd8f0d8468c75f3005",
		"uncompressedMD5": "17456aeb6568e0381c2443a44eea925b",
		"downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-pibakery.7z",
		"uncompressedSize": 4919918592,
		"filename": "raspbian-pibakery.img",
		"compressedFilename": "raspbian-pibakery.7z",
		"uncompressedFilename": "raspbian-pibakery-new.img",
		"displayName": "Raspbian Full",
		"installed": true,
		"skipVersion": 0
	},
	{
		"version": 6.0,
		"compressedMD5": "81754f93de945926c93bab3f85ef5b2e",
		"uncompressedMD5": "6300db151cf65f4f55a4f4c9bc1e2840",
		"downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-lite-pibakery.7z",
		"uncompressedSize": 1858076672,
		"filename": "raspbian-lite-pibakery.img",
		"compressedFilename": "raspbian-lite-pibakery.7z",
		"uncompressedFilename": "raspbian-lite-pibakery-new.img",
		"displayName": "Raspbian Lite",
		"installed": true,
		"skipVersion": 0
	}
]`

I don't understand image.json `"version":6.0` is correct for 
this version of raspbian-lite-pibakery.7z

> 2017-11-29-raspbian-pibakery
https://github.com/davidferguson/pibakery-raspbian/releases

I want to use raspbian-lite OS
> RASPBIAN STRETCH LITE
> Minimal image based on Debian Stretch
> Version:November 2017
> Release date:2017-11-29
> Kernel version:4.9
> SHA-256:e942b70072f2e83c446b9de6f202eb8f9692c06e7d92c343361340cc016e0c9f

would you tell me how to write  image.json for 2017-11-29-raspbian-pibakery?

any help is welcome.
thanks

davidferguson, 2018-08-31:
Just as an update: PiBakery v2 has better Linux support. You'll still need to make sure you have `kpartx` installed though, until I can figure out how to make a Linux installer.

commit 8e7c597c (2016-09-14): add linux support, fixes #2

added support to write and update SD cards on linux