davidferguson/pibakery #3: Configurable base OS image

Nzbuu, 2016-09-07:
It would be nice to be able to select "Raspbian" or "Raspbian Lite" as the base OS image to write to the SD card.

alecthegeek, 2016-09-07:
I agree. I always use Raspian "Lite" as my starting point for builds

It's easier to move the smaller images around the network plus people can use any size SD card they want

brianjking, 2016-09-07:
This would definitely be nice. I'd like the option to allow for storing the ISO files or for downloading these fresh each time PiBakery is used. 

Either way, awesome job and thank you!

japharl, 2016-09-07:
Third agree.  I also use raspbian lite, primarily as most of the time I don't have a desktop with my pi...  But I can also see this of value for "other distros" that are available...  As an alternative, it might be possible to write a block that would make the full distro a "light" distro, but then you are adding a bit of complexity there and wear & tear on the sd card...

davidferguson, 2016-09-07:
This appears to be the most requested feature for a future version of PiBakery, so I'm working on that now. The changes that need to be made to make a the Raspbian image PiBakery compatible are actually very small - see [the PiBakery-raspbian GitHub page](https://github.com/davidferguson/pibakery-raspbian) for the exact changes I made.

The main issue with adding in Raspbian Lite is that I'll need to check to make sure that all blocks work with it as well, and don't require a package that only Raspbian Full has. I'll be doing that over the next few days, and if everything goes well, a new version of PiBakery should be available in a few days to a week.

I'll probably have an option in the installer for you to select what distros(s) you want to use with PiBakery, and then the installer will download them, rather than bundling them all with the installer, as this should save users from downloading a massive file. I know the installers that I use do support this - I'll just need to work out how exactly!

And lastly thanks for your interest in PiBakery! Your support means the world to me.

Nzbuu, 2016-09-07:
I've been wondering if you're making things hard for yourself by trying to write the entire disk image from PiBakery. If the process were implemented along the lines of:

1. Write OS image to SD card
2. Install PiBakery application directly on SD card
3. Write PiBakery configuration files

then there are existing processes and tools in place for step 1, you wouldn't need to implement it in PiBakery as well and it would work automatically with any Raspberry Pi OS (I think).

However, I expect that the process is actually implemented as:

1. Download OS image
2. Install PiBakery application directly in OS image
3. Write OS image to SD card
4. Write PiBakery configuration files

Is this correct?

I certainly think that it would be better if PiBakery downloaded the Raspbian image itself, rather than including it in the installer.

davidferguson, 2016-09-07:
Currently when you download PiBakery, bundled is a modified version of Raspbian Full that [contains a few tweaks](http://github.com/davidferguson/pibakery-raspbian) to make it work with PiBakery. These tweaks have been applied manually by me.

When you write an SD card with PiBakery, it writes the modified version of Raspbian onto the SD card, and then writes some scripts to `/boot/PiBakery`. When you [edit an SD card](http://www.pibakery.org/docs/edit.html) with PiBakery, it reads the config from the `/boot/PiBakery/` directory and then saves them back there once you're finished editing.

In the next version of PiBakery, the modified raspbian.img will not be included in the installer, and instead it will download it during the installation. This allows the installer to have an option for people to choose whether they want Raspbian Full, Lite, or both.

lurch, 2016-09-07:
@Nzbuu AFAICT there's two 'halves' to PiBakery - there's the main GUI (application) 'config' side which you use to edit all the blocks etc. and this then writes out some config files to the `/boot` partition on the SD card (readable by Windows and Mac). I believe that's what https://github.com/davidferguson/pibakery contains the source for.
And then there's the 'doing stuff' side, which runs (from within the `/root` partition) at bootup, and performs various actions, based on what the config files on the `/boot` partition say. I believe that's what https://github.com/davidferguson/pibakery-raspbian contains the source code for.

Obviously without the 'doing stuff' part, then the 'config' side by itself is of very little use. The 'doing stuff' part needs to be installed inside the `/root` partition on the SD card, and neither Windows or Mac are able to read or write the `/root` partition. Therefore @davidferguson has created a special Raspbian image with the 'doing stuff' already installed, and this is both why the installer includes the complete (customised) Raspbian image, and why this won't automatically work with any other Raspberry Pi OS.

@davidferguson Please feel free to correct me if I've got any of that wrong!

davidferguson, 2016-09-07:
That's exactly right @lurch - except you can't forget that another crucial part of PiBakery are the scripts, or blocks, which can be found https://github.com/davidferguson/pibakery-blocks. They are the actual scripts that actually perform the actions, and are run by the code in https://github.com/davidferguson/pibakery-raspbian.

lurch, 2016-09-07:
Ahh, I hadn't realised that the blocks weren't in the same repo as the application :)

Nzbuu, 2016-09-07:
@davidferguson , @lurch , that's what I thought and basically matches the second process that I described. It's very clever!

Would it be possible to instead install the PiBakery files on the Raspberry Pi as part of the bootstrap script in the boot sequence. So the full process would be:

1. Write vanilla OS image to SD card, if it's blank
2. Write pibakery-raspbian and pibakery-blocks ZIP files in /boot/pibakery
3. Write bootstrap script in /boot/pibakery that installs the contents of the ZIP files in the correct locations
4. Write PiBakery configuration files in /boot/pibakery

Then, on boot, Linux would run the bootstrap script that installs the PiBakery files first and deletes itself, then PiBakery itself would run. Is this possible?

It would be great if it was possible install PiBakery without overwriting the entire contents of the SD card!

lurch, 2016-09-07:
> Then, on boot, Linux would run the bootstrap script

There's no pre-set mechanism within Raspbian that you can use to force it to run any scripts on the `/boot` partition. (at least, not without editing files in the `/root` partition first, which defeats your objective!)

Nzbuu, 2016-09-07:
> > Then, on boot, Linux would run the bootstrap script

> There's no pre-set mechanism within Raspbian that you can use to force it to run any scripts on the /boot partition. (at least, not without editing files in the /root partition first, which defeats your objective!)

That's what I thought. Presumably using the kernel "init" parameter would either not work or be dangerous. Sorry, my Linux knowledge is not good.

> > it would also avoid losing data that was already on the card.

> I guess in theory (if you were clever) you could 'apply' PiBakery to an already-booted card, and then the PiBakery application would be none the wiser, and wouldn't know that it wasn't a card that it hadn't created itself. Might be a useful script @davidferguson ? ;-)

I'm not sure what you mean here. Or do you mean modify an arbitrary OS image file to install the PiBakery "do stuff" files?

lurch, 2016-09-07:
> Presumably using the kernel "init" parameter

Dunno, that's not something I've ever experimented with.

> Or do you mean modify an arbitrary OS image file to install the PiBakery "do stuff" files?

Yup, that's exactly what I meant :-)  Along with also installing 'template' config files onto the `/boot` partition, so that the PiBakery application (which, on Windows and Mac, can only see the `/boot` partition on the SD card) is aware that this is a "pibakery" SD card.

japharl, 2016-09-07:
BTW, I'll be adding a few blocks (ideally) over the next few (time period)
because this is a cool project.

davidferguson, 2016-09-07:
@lurch 
>I guess in theory (if you were clever) you could 'apply' PiBakery to an already-booted card, and then the PiBakery application would be none the wiser, and wouldn't know that it wasn't a card that it hadn't created itself. Might be a useful script @davidferguson ? ;-)
I actually have a debian package on the way - and I'm hoping Raspberry Pi Foundation might add it to their education repo, and the you'll just be able to tun `apt-get install pibakery` on a non-PiBakery SD to get it working. That's lower down on my (ever growing) priority list, but I'll still be working on it!

mukulseth, 2016-09-13:
This is exactly what I was looking for. fantastic work! @davidferguson any update on this feature? awaiting eagerly :)

carlosperate, 2016-09-13:
> I actually have a debian package on the way - and I'm hoping Raspberry Pi Foundation might add it to their education repo, and the you'll just be able to tun apt-get install pibakery on a [...]

Maybe I do not understand the OS changes required to run the scripts from the boot partition, but I don't really see how that would help with being able to use pibakery on any raspbian image?
If you need your pibakery scripts from to the linux distribution filesystem to run on startup, and this deb package installs all that, you would still need a way to install that on raspbian on first boot no? Which would also require an internet connection. Am I misinterpreting your message?

lurch, 2016-09-13:
I think @davidferguson is suggesting adding this so that someone who has _already_ got Raspbian installed to their SD card, and doesn't want to wipe it just to try out PiBakery, can at least use PiBakery to setup the "OnNextBoot" / "OnEveryBoot" stuff. Obviously if installing PiBakery post-OS-install, you'll never be able to use the "OnFirstBoot" stuff.

As regards needing an internet connection to install PiBakery, if you use wired ethernet then no configuration is needed, the Pi has an internet connection as soon as it's booted up.

Earl-Brown, 2016-09-13:
Isn't it one of the issues with different distributions that they have different "startup" scripts/locations?

If so, would it be possible to modify piBakery to have a distinct "write" block for each supported distro?

Then the idea of "add custom startup scripts to do the piBakery work" should be easier to implement.

Of course, if some of the "do piBakery work" scripts are distribution-specific, it would probably be useful to tag the blocks with a valid distributions list.

davidferguson, 2016-09-14:
> Of course, if some of the "do piBakery work" scripts are distribution-specific, it would probably be useful to tag the blocks with a valid distributions list.

That's the approach that I'm going for right now. I'll add an extra parameter in the block JSON file that specifies what operating systems that block can be used for (although they all MUST work with Raspbian Full, to preserve backwards compatibility with older versions of PiBakery).

davidferguson, 2016-09-16:
PiBakery now supports multiple operating systems (although I've only made images for Raspbian Full and Raspbian Lite so far), if you build from source. I'll be pushing out an update which adds this into the installers very soon - basically as soon as I finish updating the installers to give you an option of which operating systems you want to install.

CleoQc, 2016-09-17:
This is fantastic! I haven't had a chance to follow the changes made to Pibakery since it came out unfortunately. But this one sounds huge to me! Great work!

SidSad, 2016-09-17:
I am struggling to get up and running with PiBakery.. Windows installer download keeps falling over so I decided to make it from source.. anyone have an example file for images.json file? The link on the source page is not yet up.

davidferguson, 2016-09-17:
@SidSad For the moment use https://gist.githubusercontent.com/davidferguson/249b28f3e41d6e9e69f8a5314320c913/raw/6b7e504f186c0f36c657eae0893bc6c6aeb8296b/images.json but this is subject to change. Can you also create a new issue about the installer failing, and tell me what errors are displayed (if any)?

SidSad, 2016-09-17:
@davidferguson thanks for sharing the example. I am trying to use it with the rasbian image from raspberrypi.org.. and I get no option for the operating system see attached snapshot.. The json file I've got is:

	{
		"version": 1.0,
		"compressedMD5": "b8e031595bacb580d5618122ae14cc39",
		"uncompressedMD5": "0d50922374043ed5246df2b316eb7d25",
		"downloadUrl": "https://downloads.raspberrypi.org/raspbian_latest",
		"uncompressedSize": 4019191808,
		"filename": "2016-05-27-raspbian-jessie.img",
		"compressedFilename": "2016-05-27-raspbian-jessie.zip",
		"uncompressedFilename": "2016-05-27-raspbian-jessie.img",
		"displayName": "Raspbian Full",
		"installed": false,
		"skipVersion": 0
	}

Updated the fields from my file with the correct MD5 checksum and everything
[image]

davidferguson, 2016-09-17:
Firstly about the weird styling for the write box - I've just fixed that now (forgot to add my .css file in the list of files to commit!).

You'll need to use the Raspbian Full and Raspbian Lite images from the [pibakery-raspbian repo page](https://github.com/davidferguson/pibakery-raspbian/releases) not the ones that you can download from raspberrypi.org. To make the images work with PiBakery I have to make some minor changes to them, so if you try and use a non-PiBakery image, nothing will happen.

And the reason nothing showed up in the list is because you need to change the `installed` parameter in the `images.json` file to `true` to tell PiBakery that you've got that installed. Apologies about that - I should have made it clearer in the instructions.

SidSad, 2016-09-17:
@davidferguson .. I thought as much but I'm getting the same "forbidden" error when I try to download the pibakery-raspbian.7z from the repo.. giving it another go now..

Thanks for the clarification on the installed=true parameter..

akotlerman, 2016-11-12:
Hi! First of all, thank you for your hard work on PiBakery! I have a few suggestions:

- Please make the "Installed=true" for images.json instruction more clear in the README.md file
- I have an issue with "installed=true" on linux. After opening the PiBakery GUI, I need to set "installed=true" again before writing to the SD card, or my img's won't show up in the list of Operating Systems. I noticed that the program automagically sets installed=false when it boots up. 
- On linux (and probably mac) its `sudo npm start`

dauuricus, 2018-01-22:
I'm in trouble at `npm start`

I downloaded raspbian-lite-pibakery.7z and set it on Pibakery `os/` directory.
when I try to start Pibakery on my Ubuntu Terminal...

here is Terminal
```

blackcat@blackcat:~/pibakery$ npm start

> PiBakery@0.3.7 start /home/blackcat/pibakery
> electron .

sh: 1: electron: not found
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn
npm ERR! PiBakery@0.3.7 start: `electron .`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the PiBakery@0.3.7 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/blackcat/.npm/_logs/2018-01-22T08_32_48_213Z-debug.log
```
and this is 2018-01-22T08_32_48_213Z-debug.log

```
0 info it worked if it ends with ok
1 verbose cli [ '/usr/local/bin/node', '/usr/local/bin/npm', 'start' ]
2 info using npm@5.6.0
3 info using node@v9.4.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle PiBakery@0.3.7~prestart: PiBakery@0.3.7
6 info lifecycle PiBakery@0.3.7~start: PiBakery@0.3.7
7 verbose lifecycle PiBakery@0.3.7~start: unsafe-perm in lifecycle true
8 verbose lifecycle PiBakery@0.3.7~start: PATH: /usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/blackcat/pibakery/node_modules/.bin:/home/blackcat/.cargo/bin:/home/blackcat/bin:/home/blackcat/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
9 verbose lifecycle PiBakery@0.3.7~start: CWD: /home/blackcat/pibakery
10 silly lifecycle PiBakery@0.3.7~start: Args: [ '-c', 'electron .' ]
11 info lifecycle PiBakery@0.3.7~start: Failed to exec start script
12 verbose stack Error: PiBakery@0.3.7 start: `electron .`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:160:13)
12 verbose stack     at maybeClose (internal/child_process.js:943:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:220:5)
13 verbose pkgid PiBakery@0.3.7
14 verbose cwd /home/blackcat/pibakery
15 verbose Linux 4.13.0-26-lowlatency
16 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
17 verbose node v9.4.0
18 verbose npm  v5.6.0
19 error file sh
20 error code ELIFECYCLE
21 error errno ENOENT
22 error syscall spawn
23 error PiBakery@0.3.7 start: `electron .`
```
in this case 19-23 are error massage.
what I do for electron finding?

and my image.json file is

[ { "version": 6.0, "compressedMD5": "cb29576520ed28fd8f0d8468c75f3005", "uncompressedMD5": "17456aeb6568e0381c2443a44eea925b", "downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-pibakery.7z", "uncompressedSize": 4919918592, "filename": "raspbian-pibakery.img", "compressedFilename": "raspbian-pibakery.7z", "uncompressedFilename": "raspbian-pibakery-new.img", "displayName": "Raspbian Full", "installed": true, "skipVersion": 0 }, { "version": 6.0, "compressedMD5": "81754f93de945926c93bab3f85ef5b2e", "uncompressedMD5": "6300db151cf65f4f55a4f4c9bc1e2840", "downloadUrl": "https://github.com/davidferguson/pibakery-raspbian/releases/download/v0.6.0/raspbian-lite-pibakery.7z", "uncompressedSize": 1858076672, "filename": "raspbian-lite-pibakery.img", "compressedFilename": "raspbian-lite-pibakery.7z", "uncompressedFilename": "raspbian-lite-pibakery-new.img", "displayName": "Raspbian Lite", "installed": true, "skipVersion": 0 } ]

I don't understand image.json "version":6.0 is correct for
this version of raspbian-lite-pibakery.7z

2017-11-29-raspbian-pibakery
https://github.com/davidferguson/pibakery-raspbian/releases

I want to use raspbian-lite OS
RASPBIAN STRETCH LITE
> Minimal image based on Debian Stretch
> Version:November 2017
> Release date:2017-11-29
> Kernel version:4.9
> SHA-256:e942b70072f2e83c446b9de6f202eb8f9692c06e7d92c343361340cc016e0c9f

would you tell me how to write image.json for 2017-11-29-raspbian-pibakery?
I downloaded raspbian-lite-pibakery.7z and set it on Pibakery `os/` directory.

any help is welcome.
thanks

dauuricus, 2018-01-22:
@lurch thank you for replying 

I omitted a few things ; )
 **npm** version is 5.6.0
and node v9.4.0
my OS is Ubuntu

according to **Installing from source**
https://github.com/davidferguson/pibakery#installing-from-source

> If you're on macOS it's important you run npm run setup the first time to create a folder inside Application Support that will hold the operating system .img files.
is that need to do on Linux OS too?

according to my .npm **2018-01-22T08_32_48_213Z-debug.log** file
`16 verbose argv "/usr/local/bin/node" "/usr/local/bin/npm" "start"
17 verbose node v9.4.0
18 verbose npm  v5.6.0`
it seems to be working npm and node.js 

I think my image.json is not macth on my raspbian-lite-pibakery.img
I just copied image.json from 

> the pibakery-raspbian repo

https://raw.githubusercontent.com/davidferguson/pibakery-raspbian/master/images.json
and pasted to my image.json.

lurch, 2018-01-22:
> Is that need to do on Linux OS too?

Nope, but the fact that `npm start` is complaining that `electron` is missing, makes me think that you forgot to run the `npm install` step. (which installs a local copy of electron into `node_modules/.bin`)

I'm afraid I don't know anything about the `images.json` part, so you'll have to wait for @davidferguson to help you with that...

dauuricus, 2018-01-22:
@lurch thank you.
electron belong to node...I did not know about node.js anything.
OK, so
Do you know how to install electron correctly on node.js ?

lurch, 2018-01-22:
[image]

:wink:

dauuricus, 2018-01-22:
Now I try `sudo npm install electron -g` on my terminal

then
```
/usr/local/bin/electron -> /usr/local/lib/node_modules/electron/cli.js

> electron@1.7.10 postinstall /usr/local/lib/node_modules/electron
> node install.js

/usr/local/lib/node_modules/electron/install.js:48
  throw err
  ^

Error: EACCES: permission denied, mkdir '/usr/local/lib/node_modules/electron/electron-tmp-download-11816-1516628013474'
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! electron@1.7.10 postinstall: `node install.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the electron@1.7.10 postinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/blackcat/.npm/_logs/2018-01-22T13_33_33_755Z-debug.log

```
OK I have to learn more about **electron** at first.

lurch, 2018-01-22:
Hmmm, why are you trying to run `sudo npm install electron -g` ??

dauuricus, 2018-01-22:
@lurch 
 I didnt know about electron , so I check electron in Japanese web programmers site.( my mother tongue is Japanese)

lurch, 2018-01-22:
Ahh. `npm install` doesn't install npm - it instead uses npm to install the additional modules that pibakery specifies that it needs (in it's `package.json`) :-)  And one of those additional modules just happens to be electron https://github.com/davidferguson/pibakery/blob/master/package.json#L40

dauuricus, 2018-01-22:
> Ahh. npm install doesn't install npm
OK

dauuricus, 2018-01-22:
> To clarify, when you run npm install in the pibakery directory, npm looks at the local package.json file, and then installs the modules that it specifies (including electron) into the local node_modules directory. No sudo needed.

OK

davidferguson, 2018-01-22:
Thanks @lurch - that's exactly right @dauuricus, if you follow the instructions under [Installing from source](https://github.com/davidferguson/pibakery#installing-from-source) that should give you a working PiBakery installation.

Please also make sure you've read and followed the [PiBakery on Linux section](https://github.com/davidferguson/pibakery#pibakery-on-linux), which states that you'll need to install the programs `kpartx` and `p7zip`. If you're on Ubuntu or Debian, these can be installed using the command: `sudo apt-get install kpartx p7zip`.

dauuricus, 2018-01-22:
@davidferguson 
thank you.
I downloaded **raspbian-lite-pibakery.7z** and extracted it.and then set `.img` file on Pibakery `os/ `directory. 
Do I need set **raspbian-pibakery.7z** on `os/ `directory too?
I want to use raspbian-lite, so I didn't downloaded **raspbian-pibakery.7z**.

> Please also make sure you've read and followed the PiBakery on Linux section, which states that you'll need to install the programs kpartx and p7zip. If you're on Ubuntu or Debian, these can be installed using the command: sudo apt-get install kpartx p7zip.

OK

davidferguson, 2018-01-22:
>I downloaded raspbian-lite-pibakery.7z and extracted it.and then set .img file on Pibakery os/directory.
>Do I need set raspbian-pibakery.7z on os/directory too?

So you've downloaded and extracted `raspbian-lite-pibakery.7z`, you then need to rename it to `raspbian-lite-pibakery.img` (rather than `raspbian-lite-pibakery-new.img`) so that it matches the `name` parameter in `images.json`.

And that *should* be it. You can run PiBakery by `cd`ing into the `pibakery` directory and then running `npm start`

dauuricus, 2018-01-22:
@davidferguson
@lurch 
; )

thank you!!
[image]

davidferguson, 2018-01-22:
No worries, thanks for using PiBakery. Please let me know here how it goes, as there ~~are possibly~~ will be some bugs in the Linux version that I need to fix.

dauuricus, 2018-01-22:
> you then need to rename it to raspbian-lite-pibakery.img (rather than raspbian-lite-pibakery-new.img) so that it matches the name parameter in images.json.

OK thanks.
[image]
I will change filename `raspbian-lite-pibakery.img` to `raspbian-lite-pibakery.img` .

lurch, 2018-01-22:
> @lurch
I install electron
npm install electron

That wasn't necessary, all you need to do (which you've now done :grin: ) is run `npm install` inside the `pibakery` directory.

dauuricus, 2018-01-22:
@lurch
@davidferguson

Oh I understand that.

lurch, 2018-01-22:
> is it work?
I feel It seems to failed in terminal

If you can actually see the PiBakery GUI, then I think it's safe to assume that it's all working okay :wink:

dauuricus, 2018-01-22:
@lurch
thank you so much !