davidferguson/pibakery #6: PiBakery Server?

dsdude123, 2016-09-07:
I would love to see some sort of PiBakery server that would allow me to push updated scripts for Every Boot and also be able to push scripts to update software from a server so that I don't have to retrive the SD card every time I want to update the scripts.

davidferguson, 2016-09-07:
That's an interesting idea, but sadly not one that I'll be working on in the immediate future, due to the volume of changes that would need to be made to the current code. I'd also need to consider the security aspect of this, as you wouldn't want unauthorised people accessing and modifying your Raspberry Pi!

lurch, 2016-09-08:
I guess @dsdude123 could write a script to download a script from the server of their choice, and then simply execute it with bash? Something like:
```bash
#!/bin/bash
MY_SERVER=http://www.whatever.com/blah
MY_SCRIPT=somescript.sh
LOCAL_DIR=/home/pi
echo "Downloading $MY_SCRIPT"
if wget $MY_SERVER/$MY_SCRIPT -O $LOCAL_DIR/$MY_SCRIPT; then
  echo "Running $MY_SCRIPT"
  bash $LOCAL_DIR/$MY_SCRIPT
else
  echo "Download failed"
  exit 1
fi
```
Note: completely untested, so use at your own risk!

(unless I've misunderstood what was being asked for?)