davidferguson/pibakery #8: produce a log of the actions

juggledad, 2016-09-07:
It would be very helpful if a log of the actions could be produced.

(for now, you can edit the script and do a gobal change of
`>/dev/null`
to
`>>/home/pi/pibaker.log`

NOTE: you will have to re-do this if you change the script and do an update in the 'bakery'

davidferguson, 2016-09-14:
I'll be adding this into the nest PiBakery Raspbian update - any thoughts to where the log should be stored? The log will probably be in `/boot/PiBakery/pibakery.log` so it is accessible via the SD card.

juggledad, 2016-09-14:
I'd just put it in the home folder of the ‘pi' user. You will be logged into the pi when you go look for it and it will be easy for noobs to find it there.

davidferguson, 2016-09-14:
The reason I though of putting it in `/boot/` is because the if you're using your Pi headless (as many PiBakery users are doing) then you can read the log file from the SD card on any computer.

juggledad, 2016-09-14:
But if you're running headless then you probably have SSH into the pi and that will drop you into to the home folder. But either way will work.

davidferguson, 2016-09-17:
I've now added this into the PiBakery source, and plan to update the installers to include this over the weekend.

The log files it writes are located:
- `/boot/PiBakery/firstboot.log`
- `/boot/PiBakery/everyboot.log`
- `/boot/PiBakery/nextboot.log`

closed 2016-09-17 by davidferguson