davidferguson/pibakery #9: Cron job support

adrienemery, 2016-09-07:
The ability to support cron job configuration would be really useful. Is there any plan for this (or maybe I missed it).

Thanks for the great work on this!

davidferguson, 2016-09-07:
It's not one that I'd though of, but I can certainly add it in! I can't garentee when though, as I have a few block suggestions and bug fixes to get through first.

However, if you want to create and submit a block for it, then have a look at the [Contributing Guide](http://www.pibakery.org/docs/contribute.html) on the PiBakery website.

davidferguson, 2016-09-17:
This has now been committed into the [pibakery-blocks repo](https://github.com/davidferguson/pibakery-blocks). I plan on pushing out a block update this weekend, which will include this block.

closed 2016-09-17 by davidferguson

adrienemery, 2016-09-19:
thank you. PiBakery is awesome!