davidferguson/pibakery #12: Read Only Root Filesystem

ekapujiw2002, 2016-09-08:
Hello,
Nice configuration tool. If you don't mind please add the option to make root filesystem read only, so no abrupt power loss can damage the os.
Thank you

davidferguson, 2016-09-14:
That's a good idea, I'll look into it. The one issue there might be with that is that from then on the SD card wouldn't work with PiBakery because it would be read only. I'll have to have a think to see if there's a way round that.

ekapujiw2002, 2016-09-14:
What i have done using arch linux arm to achieve read only system is like below,maybe you can add it to the recipe :
1. Make a read write partition to store all the required configuration file and make a symlink to the original path
2. Disable swap
3. Make the partitiion can be mounted read only or read write at any time. By default the system is read only.
4. Disable all log,or redirect log to memory with limited amount (i usually used 16MB)
Hope that will give you some hint
Works very well for me and no worry at all on power loss