davidferguson/pibakery #15: Generic Rasbian image support

novirium, 2016-09-08:
Is there any opportunity to use something like https://github.com/BytemarkHosting/pi-init2 to start your scripts, rather than have to use a custom rasbian image? That way the only modifications needed are to /boot, and users could supply their own image.

davidferguson, 2016-09-08:
That's very interesting - I've never heard of pi-init2 before. I'll have a look at it, and see if I could add it in, although to keep backwards compatibility with previous versions of PiBakery, I might not use this for now.

Thanks for letting me know about this!

lurch, 2016-09-08:
I had a quick look at that, and https://github.com/BytemarkHosting/pi-init2/blob/master/src/projects.bytemark.co.uk/pi-init2/init.go#L53 is a bit concerning:
> // for now we don't care about permissions

It seems to delete files from the `/root` partition and replace them with symlinks to files on the `/boot` partition; and as the boot partition is FAT (which doesn't support file permissions) then the symlinks won't support file permissions either.

It also seems to assume a 'virgin' `cmdline.txt` that the user hasn't modified (and simply overwrites it).

Still, interesting... (it uses the approach suggested by https://github.com/davidferguson/pibakery/issues/3#issuecomment-245340567 )

davidferguson, 2016-09-08:
>It also seems to assume a 'virgin' cmdline.txt that the user hasn't modified (and simply overwrites it).

That's what grabbed my attention when I looked at it, and put me off a little bit. Of course it wouldn't be difficult to change that, but for the moment I think I'll stick with the method I use. That's not to say I won't use pi-init2 in the future, though!

davidferguson, 2018-06-29:
This has been implemented in PiBakery v2, see https://github.com/davidferguson/pibakery/issues/3#issuecomment-401478400

closed 2018-06-29 by davidferguson