davidferguson/pibakery #16: Save image to file

magicgg91, 2016-09-08:
Hello,
I think it will be interresting to have the choice to write imge to file or onto SDcard.
Thanks

davidferguson, 2016-09-08:
Unfortunately due to the way PiBakery works, this won't be possible (at least on Windows).

When you click "Write" in PiBakery, a modified version of Raspbian.img is written to the SD card. When that write is complete, PiBakery then generates onboot scripts from your blocks, and writes them to the (now mounted) `/boot/PiBakery/` directory.

On Linux and Mac, I could have the modified Raspbian.img written to a loopback device instead of the SD card, and then edit the files off that, but as far as I know, there is no way to do this programatically on Windows.

lurch, 2016-09-08:
Just to clarify: when @davidferguson says "a modified version of Raspbian.img" he means a pre-modified (by him) version of Raspbian.img already included with the installer, **not** a version of Raspbian.img that gets modified by PiBakery at run-time ;-)

@davidferguson I believe it would be possible on Linux, but not on Mac, as Mac doesn't include the appropriate ext4 filesystem driver?
For ease of maintenance, IMHO it'd be better to keep the Mac and Linux versions working the same as the Windows version. (which means this issue should probably be closed as "wontfix" ?)

davidferguson, 2016-09-08:
>but not on Mac, as Mac doesn't include the appropriate ext4 filesystem driver?

That's probably true - I've not actually tested it on Mac.

closed 2016-09-08 by davidferguson

freerobby, 2018-08-13:
@davidferguson Would you consider reopening this issue for Pi Bakery v2? Correct me if I'm wrong, but it would seem to be a much simpler implementation, as you would not need to muck with the ext4 system at all.

davidferguson, 2018-08-31:
Unfortunately this is still not possible, as it requires modifying the fat32 partition of the image. The way I currently do this is to write the image, mount the SD card, and make the changes. I am not aware of any way to edit the fat32 partition of the image without writing it to the SD card first. If there is a way, let me know and I can consider it.