davidferguson/pibakery #21: Importing new blocks does not work

jhergeth, 2016-09-11:
I downloaded the current version of pibackery, followed the descriptions in http://www.pibakery.org/docs/contribute.html and tried to create a new block to install java on the raspberry.
Importing my new block by drag/dropping it on pibakery does not work. I always get the "forbidden" mouse cursor and nothing happens:
[image]

Any suggestions?

With regards
jhergeth

davidferguson, 2016-09-11:
From your cursor, I presume you're running Windows - could I ask what version you're running?

I've actually already released a workaround for this - but I've neglected to update my docs - apologies about this! You can use the shortcut `ctrl shift +` to open a folder dialog chooser where you can select your block folder to import it that way.

Thanks for pointing this out to me.

jhergeth, 2016-09-12:
I use W10. 64 Bit, German

davidferguson, 2016-09-12:
Did the `ctrl shift +` shortcut work for you?

jhergeth, 2016-09-12:
Yes!

Currently testing the new block (Oracle-JDK install)

davidferguson, 2016-09-12:
Just for others who stumble across this thread - the reason drag-drop doesn't work on Windows is because Windows doesn't allow drag-drop between elevated and non-elevated programs. As PiBakery is elevated, but explorer is not, you can't drag between them.

Have a read of [this Microsoft article](https://blogs.msdn.microsoft.com/patricka/2010/01/28/q-why-doesnt-drag-and-drop-work-when-my-application-is-running-elevated-a-mandatory-integrity-control-and-uipi/) for more information.

jhergeth, 2016-09-17:
It seems that the lines:
```
	electronLocalshortcut.register(mainWindow, 'CommandOrControl+Shift+Plus', function()
	{
		mainWindow.webContents.send('testBlock');
	});
```
have been lost in the new version which supports also rasbian lite...

davidferguson, 2016-09-17:
Thanks @jhergeth - not quite sure what happened there! I'll add it back in.

commit a780f4e3 (2016-09-17): added back in import block shortcut, fixes #21

The shortcut that import the test block has accidentally been removed.
This fix adds it back in again so you can use the shortcut
`CommandOrControl+Shift+Plus` to import blocks.
closed 2016-09-17 by davidferguson