davidferguson/pibakery #24: Windows 10 endless script hosts

MonkoftheFunk, 2016-09-12:
After finished loading, it endlessly spawns microsoft console based script host, console window host, windows command proccessor. The app works fine just quickly uses up my computer's resources.

davidferguson, 2016-09-12:
That's unusual - I've never had that before. With the PiBakery window selected, you can use the shortcut ctrl-shift-i to open the Developer Inspector. Could you choose the console tab at the top of the Inspector and then tell me if there are any errors in there?

davidferguson, 2016-09-13:
@MonkoftheFunk Could you also tell me if you're asked to grant PiBakery admin privileges?

MonkoftheFunk, 2016-09-13:
In console looks like it tried to load D:\PiBakery\blocks.xml and E:\PiBakery\blocks.xml and it failed, not sure if that's what is causing it. I believe the js line was 1057
Yes PiBakery did prompt for admin privileges and it was granted.

davidferguson, 2016-09-13:
Any errors about `blocks.xml` are normal - they're just for the updating part of PiBakery, where they check to see if that disk is a PiBakery SD card.

Would you be able to attach a screenshot of what happens?

MonkoftheFunk, 2016-09-14:
Sure, though it isn't very telling showing all the duplicate process. I'm also going to try it on 2 more computers to test. Perhaps you might have a version with some console logs? I guess I could step through the JavaScript to see what is looping. Any ideas what would use the windows script host?

davidferguson, 2016-09-14:
The only two places that PiBakery spawns a new process are:

- When PiBakery first launches, it check to see if it's running as admin, and if not, tries to elevate itself by reopening itself under admin privileges.

- When you click `Write`, it creates a new process of [CommandLineDiskImager](https://github.com/davidferguson/CommandLineDiskImager) - my version of Win32DiskImager

Based on the fact that the windows are showing as soon as you open PiBakery, I'm guessing it's the part that's supposed to elevate PiBakery by reopening itself as admin. Sadly I don't have a Windows 10 machine to experiment on, but I'll have a look in the code (it's handled by [elevate.js](https://github.com/davidferguson/pibakery/blob/master/elevate.js)) and see if I can spot anything.

I should have asked earlier, but do the console windows have anything inside of them? Error messages, log info, etc?

MonkoftheFunk, 2016-09-15:
So it also happens on my other computer. It only happens after it has finally loaded (no spinner). They are background processes so no display.

MonkoftheFunk, 2016-09-15:
Perhaps I'll take a look and see if windows logged anything

MonkoftheFunk, 2016-09-16:
Could this be the reason:
https://github.com/keystonejs/keystone/issues/145

For some reason the js files are using Windows host instead of node?

MonkoftheFunk, 2016-09-19:
So the amount of process isn't "endless" around 200 but then it shrinks back  and grows again in waves. No real apparent reason, Tool still works on a better machine. Still think some how it's running js files through windows script host instead of node, but can't understand that it must be still using node since it doesn't happen until after the blocks are loaded

tranber, 2016-11-04:
Sorry, I have reported similar issues in another bug tracker:
   https://github.com/davidferguson/pibakery-raspbian/issues/12
might be not the correct one.

I've looked quickly at the code and I am wondering if the 'drivelistScanner' could be linked to that? it is run every second?
It turns out that I was unable to eject my SD card, even if I had quited PiBaekry, until I finally kill all the pending processes.

tranber, 2016-11-04:
I have experimented changing the 'interval' parameter for the 'drivelistScanner', setting it to 20000 (20s?) and... it solves my issue: no more hundred of processes!
I guess that my machine might be slow to answer (>1s), and if a drivelist request starts while the previous one is not over, maybe it causes the problem.
I don't know if there is something more elegant that raising this value, at something like 5 seconds? and what would be the impact...
(Disclaimer: I don't know node.js, so I am just guessing...)

davidferguson, 2016-11-05:
Thanks so much for your work on this - I've just tried running PiBakery on an older machine and reproduced the error exactly how you described. For the moment, I'll probably reduce the interval which means that the popup you get when you insert an existing SD won't be so fluid and immediate, but this is less important that actually getting PiBakery to run!

tranber, 2016-11-06:
Great, I think it is a good compromise!

MonkoftheFunk, 2016-11-10:
Thanks guys!

HawkeyeCAP, 2017-03-25:
Major noob here, trying to eliminate this same issue. 

How does one change the 'interval' parameter for drivelistScanner?

What file do I open? Editor? 

Sorry, I am totally new at this, I tend to play with other things.

tranber, 2017-03-26:
I would suggest the following:
in the file resources\app\app\pibakery.js
around line 890
change the parameter interval of the call to drivelistScanner.
I have this here:
`var scanner = new drivelistScanner({interval: 5000})`
the value 5000 (5 seconds) seems OK for me.

HawkeyeCAP, 2017-04-01:
Thank you very much. 

Tried the five seconds, it still lagged a bit.  Changed it to eight, it worked fine. 

Been able to use it now, for three different Pis, and it is very handy. Great program, just took a little tweaking. But, hey that's part of the fun.

Kudos to Mr. Ferguson.

davidferguson, 2018-06-29:
This was to do with a bug in **drivelist-scanner**. As this module is no longer used in PiBakery v2, this should no longer be an issue.

closed 2018-06-29 by davidferguson

maksim3773maks, 2019-02-23:
Hi, after changing the parameters -interval of the call to drivelistScanner / this did not solve my problem
Maybe I did something wrong in the code?
I'm not good at this, thanks))
[image]