davidferguson/pibakery #26: wifi setup needs to support setting country

vinceskahan, 2016-09-12:
The wifi setup doesn't work for folks outside GB.  Need to support the country=XX line that goes at the top of the wpa_supplicant file

vinceskahan, 2016-09-12:
> Thanks for letting me know about this - I actually wasn't aware of the
> country paramater in wpa_supplicant!
>
> Do you think that the best option for this would be to have a drop-down
> menu on the wifi block for the user to select their country?

I don't know what your interface should be, I'm basically looking at your
code for how to do similar things CLI only without scratch.    Sounds
reasonable though.

You might also look into items to set the locale(s) and keyboard settings
and timezone.  Being in the US I always need to set those things vs.
vanilla Raspbian (plus the wifi country code).

chrishoage, 2016-09-12:
FWIW country does not seem to be _required_ from what I can tell. 

I removed the `country` line and reconfigured and everything connected and worked just fine.

davidferguson, 2016-09-13:
@vinceskahan Blocks for keyboard layout and timezone should also be on the way.

@chrishoage Can I ask what country you're in? I know you don't need the `country=` line in the UK, but I'm not sure about other countries.

spuder, 2016-09-13:
The wireless regulatory domain is [burned into the rom of most wireless devices](http://kernelmag.dailydot.com/features/report/8051/the-mystery-of-wifi-channel-14/). 

Setting the regulatory domain to asia will unlock wifi channel 13. Setting it to japan will unlock 13 and 14.

vinceskahan, 2016-09-13:
Be a little careful with how you approach this one. The country code line
is set to GB at the top of the default wpa_supplicant.conf file as
delivered in Raspbian.  I have to assume they did that for some reason.  No
clue what the wifi chip in a pi3 defaults to, for example.

Suggest editing the line in wpa_supplicant.conf and putting the right value
in, not simply hoping deleting it works for everybody regardless of where
they are.

chrishoage, 2016-09-13:
@davidferguson I am in the US, and the config file worked with out `country` set at all. Not sure if it was a fluke or not, just wanted to share the info. 

The only reason I tried was I looked at some default `wpa_supplicant.conf` configs and the `country` was commented out.

Barry4679, 2016-09-16:
FYI: current version establishes WiFi comms successfully in Australia.

haiqu, 2016-10-26:
Whether it works with UK specified is not the point. Each country has specific channels that can be used and no others. If this isn't set correctly you're operating illegally, and could possibly be interfering with users on adjacent bands.

eddwatts, 2016-11-01:
I have been working on this myself with an updated block - have submitted a pull request to update it but if you want to look what iv done:
https://github.com/eddwatts/pibakery/tree/master/pibakery-blocks/wifisetup

davidferguson, 2018-03-14:
Sorry this took so long coming (the wifi country code can now be set by a text field in the `wifisetup` block).

Today's release of Raspbian (which is required by the Pi 3+) forces the country code to be set, and without it you can't use the wifi interface.

I've gone with a text field over a dropdown menu because I found that on some computers Blockly wouldn't display such a large menu correctly, and therefore a text field would be the next best thing.

closed 2018-03-14 by davidferguson