davidferguson/pibakery #27: PiBakery won't open (windows 10)

UberFluff, 2016-09-12:
I'm on windows 10 and i tried to install and launch PiBakery, but it won't launch and electron keeps creating new process i used both the installer and the source code with git (I'm on version 0.2.4)

UberFluff

davidferguson, 2016-09-12:
This has actually already been reported in #24 but I'm still no further into solving it.

Does the [PiBakery window](http://pibakery.org/img/blocks-on-workspace.png) actually open, or do you just get many console windows? Would you be able to share a screenshot of what happens?

Thanks

UberFluff, 2016-09-12:
Sorry for the double post, I wasn't sure it was the same :/ ...
No, there's nothing at all, just the little loader above my mouse and when I check in my process there's pibakery opening and closing endlessly. Can't take a screenshot for the moment but it wouldn't change anything, since nothing appear, the pibakery process just use a little bit of RAM and nothing else, no CPU, no disk

davidferguson, 2016-09-12:
When you try and run it from source with npm start, does it give any error messages in the console?

UberFluff, 2016-09-13:
No, it downloads all the modules and when i type "npm start" it gives me this : 

C:\Users\lucas\Downloads\pibakery>npm start

> PiBakery@0.2.4 start C:\Users\lucas\Downloads\pibakery
> electron .

C:\Users\lucas\Downloads\pibakery>

closed 2018-01-12 by UberFluff