davidferguson/pibakery #28: License request

davidsharp, 2016-09-13:
I noticed this project was missing a license, as I was curious if/how I could use this as part of a larger project and I don't know where I would stand with such a thing.

davidsharp, 2016-09-13:
Actually, I've just noticed the license in the `main.js`.

A LICENSE.md would still be nice.

davidferguson, 2016-09-13:
I've got it as [COPYING](https://github.com/davidferguson/pibakery/blob/master/COPYING), which I've seen in a few other projects.

I've just looked at https://www.gnu.org/licenses/gpl-howto and it recommends using COPYING, so unless there's an issue with using that, that's what I'll stick with.

closed 2016-09-13 by davidferguson

carlosperate, 2016-09-13:
Looks like github only picks up the License to display on the front page if using the `LICENSE` filename. 
https://help.github.com/articles/adding-a-license-to-a-repository/

Not a good enough reason to rename your file, but I though it'd be interesting to share. I woudn't be surprise if a future update also supports `COPYING`.