davidferguson/pibakery #29: Mac OS X El Capitain - not starting

mukulseth, 2016-09-13:
Hi,

I've been trying since yesterday to start up the app from source. But it seems to get stuck on "Attempting Auto-Fix..." I'm attaching some screenshots to show the error messages and the flow.

[image]
[image]
[image]
[image]

I did grant permission to the app when it starts up initially. Not sure what it could be. Also, there are no error messages in the console.

mukulseth, 2016-09-13:
I've tried a few times to ensure I'm following the instructions exactly.. I'll double check to make sure the version of node isn't causing this problem

mukulseth, 2016-09-13:
These logs from the syslog might be relevant -- 

=======
Sep 13 15:57:50 prime-5 launchservicesd[85]: SecTaskLoadEntitlements failed error=22
Sep 13 15:57:50 prime-5 lsd[296]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Sep 13 15:57:50 prime-5 launchservicesd[85]: SecTaskLoadEntitlements failed error=22
Sep 13 15:57:50 --- last message repeated 1 time ---
Sep 13 15:57:50 prime-5 appleeventsd[56]: SecTaskLoadEntitlements failed error=22
Sep 13 15:57:50 prime-5 launchservicesd[85]: SecTaskLoadEntitlements failed error=22
Sep 13 15:57:50 prime-5 sudo[3298]:    mseth : TTY=ttys003 ; PWD=/Users/mseth/Projects/physi/lg-link-app ; USER=root ; COMMAND=/Users/mseth/n/lib/node_modules/electron/dist/Electron.app/Contents/MacOS/Electron .
Sep 13 15:57:50 prime-5 Electron Helper[3297]: GVA info: preferred scaler idx 0
Sep 13 15:57:50 prime-5 lsd[213]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Sep 13 15:57:50 prime-5 appleeventsd[56]: SecTaskLoadEntitlements failed error=22
Sep 13 15:57:50 prime-5 Electron Helper[3300]: GVA info: preferred scaler idx 0
Sep 13 15:57:50 prime-5 lsd[213]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Sep 13 15:57:52 prime-5 kernel[0]: Sandbox: diskutil(3309) System Policy: allow(0) file-ioctl /dev/rdisk0
Sep 13 15:58:03 prime-5 mds[66]: (DiskStore.Normal:2382) 5001 1.000005
Sep 13 15:58:18 prime-5 lsd[213]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Sep 13 15:58:18 prime-5 lsd[296]: LaunchServices: Could not store lsd-identifiers file at /private/var/db/lsd/com.apple.lsdschemes.plist
Sep 13 15:58:40 --- last message repeated 2 times ---

davidferguson, 2016-09-13:
It seems there is a bug when you try and built from source. I think that PiBakery is not managing to extract the raspbian 7z file which it downloads, and therefore every time has the error of not being able to find raspbian.img.

If you go to the folder `/Library/Application Support/` is there a `PiBakery` folder in there? If there is, could you post the output of its contents?

mukulseth, 2016-09-13:
@davidferguson 

I'm not even sure it had a chance to download the file.. it fails very quickly. And there doesn't seem to be a `PiBakery` folder in `/Library/Application Support/`

Where would the compressed 7z file be downloaded?

The Terminal titlebar seems to cycle repeatedly (very fast) between `awk` and `diskutil` while it's stuck on "Attempting Auto-Fix" -- no messages in the console though

davidferguson, 2016-09-13:
The issue might very well be that you need to manually create the `PiBakery` folder inside `/Library/Application Support/` (the installer usually does this for you). Try creating it, and see if the Auto-Fix gets any further ahead (I fear that there might still be a bug in the extraction process).

mukulseth, 2016-09-13:
I created the folder.. there doesn't seem to be any change :(

marcelogaio, 2016-09-14:
Same problem here. OSX El Capitan, identical issue.

davidferguson, 2016-09-14:
Whilst I try and fix this, I'd recommend just using the Mac installer. If you want to edit the files from the installer, they can be found in `/Applications/PiBakery.app/Contents/Resources/app/` and the raspbian image can be found in `/Library/Application Support/PiBakery/`

davidferguson, 2016-09-14:
I've updated my README to include that you need to create the `/Library/Application Support/PiBakery/os` directory before running `npm start`, and I've just checked it on my Mac which also runs El Capitan, and it now works (although the progress bar does stay on nearly complete for a few minutes - just wait for it, it should complete).

Please let me know if that works for you.

mukulseth, 2016-09-14:
@davidferguson yep.. that seems to fix the issue :) thanks!

davidferguson, 2016-09-14:
Great to hear, @mukulseth!

closed 2016-09-14 by davidferguson