davidferguson/pibakery #31: Other distros

deavmi, 2016-09-14:
Could you get this script to work with something like the Ubuntu MATE Pi version?

davidferguson, 2016-09-14:
Distros based off debian (Raspbian Full, Raspbian Lite, Ubuntu) should be able to support PiBakery, and in the next update I'm hoping to have Raspbian Lite support. I'll look into Ubuntu MATE as I've not used that before, but in the near future it should be supported.

jhergeth, 2016-09-14:
That's nice! I definitly am interested in the Rasbian Lite version!

unixfox, 2016-09-14:
It could be nice if you support ArchLinux with netctl for networking !

davidferguson, 2016-09-14:
I've never actually used Arch Linux before, so I'll need to investigate further before saying it would be possible. It would probably mean that many of the blocks would need to be rewritten, but if I (or somebody else) does that then adding in PiBakery should be simple, as both Arch and Debian use Systemd.

deavmi, 2016-09-14:
Awesome :)

Phrohdoh, 2016-10-01:
It would be great to support http://www.raspberry-asterisk.org/.

If you decide to pursue that and would like assistance please don't hesitate to ping me.

deavmi, 2016-10-01:
Asterisk sounds cool.

simon-czychy, 2016-10-08:
Kali linux would be nice too :)

deavmi, 2016-10-08:
Yeah

agsdot, 2016-12-02:
Any updates on this? @unixfox and @davidferguson, any insights on how to get archlinuxarm working with this project?

Also, @unixfox, following directions on [https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3](https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3) in the installation tab, how would you make an image suitable for something like etcher.io much less pibakery?

jamiedarcher, 2018-01-09:
I second the Kali distro

davidferguson, 2018-01-10:
Firstly sorry for not responding to this thread for over a year - hopefully this extra long reply makes up for it!

----

Supporting non-Raspbian operating systems comes with the issue that I'd need to test (or get others to help test) every block with every platform, and then implement a system of disabling certain blocks if they're not compatible with a chosen platform.

However currently even testing every block with just Raspbian and Raspbian Lite (the two releases that I currently have with PiBakery) takes a bit of time, and is the main reason why PiBakery often lags behind Raspbian updates.

----

Additionally, the way that the Windows and Mac installers currently work is that they have Raspbian and Raspbian Lite bundled with them, which the user can choose to, or not to, install. Obviously as I add additional operating systems the size of the installer grows, which is undesirable.

The solution to this would be to have the installer download the chosen operating systems during install rather than bundling them all together - and while both the Windows Installer tool and Apple's **Installer.app** support this, there is a really annoying bug in **Installer.app** where the percentages for the download are wrong - they show the percentage x100. It's probably not a deal breaker, but there may also be an issue with **Installer.app** and signing packages - I'll need to look into this in more detail.

Suffice to say, it may be likely that I have to write my own installer app for Mac, or have PiBakery ask what images to install the first time you open the app. Because of this additional complexity, additional images aren't overly high on my priority list, and when you add in the fact of having to test all the blocks as described above, it's not something that I'll probably be adding for a while.

----

Basically what I'd need to be able to support non-Raspbian operating systems are people that would be willing to test these OSses with new blocks when a new release of that OS is released.

I'll be looking into the issue with Apple's **Installer.app** as I intend to move the installation process to the downloading-images-on-the-fly type method for the next major release of PiBakery anyway. I'll update this thread with progress on this.

----

###### Note to @agsdot: The instructions in the link for Arch are showing how to write the Arch OS straight onto a SD card. (Note, they're Linux only.) It should be fairly trivial to modify them so that it creates a .img file - which can be incorporated into PiBakery.

jamiedarcher, 2018-01-19:
I'd be happy to be a tester

davidferguson, 2018-06-29:
Now that PiBakery v2 supports using any Raspbian .img file (and no longer bundles the images with the app), it should be possible to use other distros too. The 'magic' installers for PiBakery are in https://github.com/davidferguson/pibakery/tree/master/resources and https://github.com/davidferguson/pibakery/tree/master/pibakery-raspbian - these are currently written for Debian/Raspbian only, but if these were modified accordingly, there's no reason PiBakery wouldn't work on other distros.