davidferguson/pibakery #32: Change ssh host keys upon first boot on a new image

M0JFP, 2016-09-14:
Hello, I really love this project and think it will be used a lot and there will be a million new blocks added as time goes on. Well done.

I create and distribute pi image in the Ham radio community, so I build the image and add the custom software and configuration, which is all well and good.

But to try and minimize the security risk as most user will place the pi on the DMZ.

Copying the image also copies the public and private keys, so anyone with a copy of the image could potentially ssh as root to the next pi with the same image!!!!

Could you add a block to change thehost keys upon first boot up:

Step # 1: Delete old ssh host keys

Login as the root and type the following command to delete files on your SSHD server:
# /bin/rm -v /etc/ssh/ssh_host_*

Step # 2: Reconfigure OpenSSH Server

Now create a new set of keys on your SSHD server, enter:
# dpkg-reconfigure openssh-server

Sample output:

Creating SSH2 RSA key; this may take some time ...
Creating SSH2 DSA key; this may take some time ...
Restarting OpenBSD Secure Shell server: sshd.

I think this feature should really be a default setting to stop any mishaps in the future...

many thanks

James Preece (M0JFP)

davidferguson, 2016-09-14:
PiBakery is actually directly taken from the official Raspberry Pi Raspbian image, which to my knowledge generates ssh keys on first boot (and therefore PiBakery should do the same). I'm not able to test right now, but I'll check it later.

From memory, PiBakery is generating new ssh keys on first boot, as I remember having to remove my stored keys on my Mac before I was able to connect to to it again. But I haven't confirmed this, and as I said, I'll check this later (or if you're able to check it as well, that would be great).

M0JFP, 2016-09-14:
I understand if you edit (wit bakery)  a fresh img from raspian the keys would be generated at that time.

However, one additional use for your software is to help customize an existing image, if that image has already been run, as I said my Ham radio image is installed on my pi, then we add the packages and customize the desktop. when I share this image, they all have the same key's.

I could just add a script to do this, but thought it would be great to add into your project and also to alert users of the potential security flaw..

Thanks very much

James Preece (M0JFP)

davidferguson, 2016-09-14:
Oh, I understand. Sorry about the confusion!

I'd be very happy to add a block to regenerate the ssh host keys - as it's a fairly trivial block, it should make it into the next block update, set to be released in the next few days.

M0JFP, 2016-09-14:
Fantastic, many thanks David!

davidferguson, 2016-09-16:
I've now added this block into the [pibakery-blocks repo](https://github.com/davidferguson/pibakery-blocks), and will push out a block update that contains this block in the next few days.

Thanks for using PiBakery!

closed 2016-09-16 by davidferguson