davidferguson/pibakery #33: HTML error 406 from Download File block.

Barry4679, 2016-09-16:
I am unsure about the url requirements or restrictions for a Download File block.

http://albumplays.com/download/test-dl/?wpdmdl=3713 

That url downloads a text file if entered into the RPi's browser. But in a DownLoad File block it returns Errror 406.  Is this a bug, or am I trying to use the block with a url which doesn't meet requirements?

[image]

Barry4679, 2016-09-16:
Further information.

This from a FAQ at my ISP

"What is the 406 Error?
406 error - Not acceptable Web browsers make a request for information from the server. When this happens, it sends an Accept header. This tells the server in what formats the browser can accept the data. If the server cannot send data in a format requested in the Accept header, the server sends the 406 Not Acceptable error.
The error can also be generated by the mod_security module. Mod_security, a type of firewall program that runs on Apache web server, scans for violations of the rules it has set. If an action occurs that violates one of these rules, the server will throw a 406 error."

Mod_security is enabled by default on hosting accounts. It cannot be set off in a shared hosting situation as it introduces security risks.

The other advice is change the request's user agent to match something expected by the server.

Looking at your code in GIT I see that you are using curl.

using curl on the RPi I get the following results. The 1st one gives a 406 error, and the second successfully returns the text file:

		curl http://albumplays.com/download/test-dl/?wpdmdl=3713
		curl http://albumplays.com/download/test-dl/?wpdmdl=3713 -A "Mozilla"

davidferguson, 2016-09-16:
Thanks for reporting this, and investigating further. I've just checked myself, and it seems to be rejecting the default curl useragent but it allows anything else.

I'm not entirely sure the legal ramifications of pretending to be a browser (I'm guessing they're negligible) but to be on the safe side I'll change the useragent to "PiBakery".

davidferguson, 2016-09-16:
I've now fixed it in the [pibakery-blocks repo](https://github.com/davidferguson/pibakery-blocks) and I should be pushing out a block update (which will include this fix) in the next few days.

Thanks for using PiBakery!

closed 2016-09-16 by davidferguson

Barry4679, 2016-09-16:
thanks for fixing. 

> I'm not entirely sure the legal ramifications of pretending to be a browser

I looked into this. They all do it. 
"and all browsers pretended to be Mozilla, and Chrome called itself Mozilla/5.0 (Windows; U; Windows NT 5.1; en-US) AppleWebKit/525.13 (KHTML, like Gecko) Chrome/0.2.149.27 Safari/525.13, and the user agent string was a complete mess, and near useless, and everyone pretended to be everyone else, and confusion abounded."
http://webaim.org/blog/user-agent-string-history/ 

I don't suppose it matters what you use, unless some uptight server somewhere validates it, although the tongue-in-cheek blog above makes this seem unlikely I suppose.

You may consider using one of the curl agent strings I found here
http://www.useragentstring.com/pages/useragentstring.php?name=cURL

davidferguson, 2016-09-16:
Thanks for that link - it was a fascinating and entertaining read! I've had a look, and some sites respond with errors if `curl` is present in the useragent, so I'll probably just keep it as `PiBakery` for the moment.