davidferguson/pibakery #37: Windows Installer Download

SidSad, 2016-09-17:
The windows installer fails stating error: "Forbidden".

Using Chrome and Windows 7 Profession (64-bit)

davidferguson, 2016-09-17:
Did the installer ask to be run as Administrator?
How much free space on your HD do you have?
Did the installer manage to open and the install begin, or did it just show an error as soon as you tried to open it?

SidSad, 2016-09-17:
It actually doesn't get that far.. the download stops midway through with the error "forbidden"

davidferguson, 2016-09-17:
So it's the actual download that fails? I've actually had some other reports of the "forbidden" error.

I'll investigate, and hopefully be able to provide a mirror download at some point.

davidferguson, 2016-09-17:
I've just put a copy of the PiBakery.Setup.exe on my server, which you can get at http://pibakery.org/downloads/

Hopefully this will help whilst I try and find out if GitHub are having issues at the moment.

SidSad, 2016-09-17:
@davidferguson thanks for the new mirror.. It downloaded fine but the card write is still not working from the windows installer but I got the rasbian image from the package and used it with the pibakery created from source.. It still doesn't seem to be working (tried setting up wifi and VNC server on first boot) though it gives the message "Write Successful".. The windows installer package doesn't contain the image.json file

davidferguson, 2016-09-17:
Yes, the installer is for the old version of PiBakery (well, it's actually the current stable version) which doesn't support multiple operating systems, and uses info.json rather than images.json.

Can you tell me what didn't work when writing to an SD from the Windows installer version?

SidSad, 2016-09-18:
The operating system was written no problem but the hostname configuration, Wifi Setup and VNC server don't seem to be working.. just gave it a second try same results

newhope, 2016-09-20:
Not only windows installer, It's also happened on Mac OS installer.
Seem those big files are served via amazon ws and it just keeps stop transferring in the middle with "network error" status, when clicking on retry and resume download, the "Forbidden" status showed up.

Tried with chrome, safari & 3rd party downloader, all fail at the middle.

I think this issue should be escalated to amazon ws?

p/s: I'm in Singapore now, just in case it may only happens on some network lines.

davidferguson, 2016-09-20:
I've actually just had this issue myself now - so I don't think it's because of network lines, it looks much more likely to be either a GitHub config issue, or AWS.

As I have no control over either of those, the only thing I can do is mirror the files on my own server and use those instead. I'm copying the files over right now, and once they're done, they'll be online at http://pibakery.org/downloads/ - but I'll change the download links on pibakery.org to reflect these changes as well.

SidSad, 2016-09-20:
Hi David any luck on the Wi-Fi setup and VNC? I'm using this on pi zero don't know if that affects anything

davidferguson, 2016-09-20:
@SidSad I've found the bug, and I'm uploading the new version right now. If all goes well it'll be available as a block update today.

SidSad, 2016-09-20:
Awesome can't wait to get my hands on it.. thanks :)

closed 2016-09-21 by davidferguson

andrewjw23, 2018-08-20:
This still seems to be a problem.  I have tried many times to download the windows installer.  Everytime the download is predicted to take many hours, but fails part way through.
Pity, Pibakery seemed like a useful tool.

davidferguson, 2018-08-20:
The download is hosted by GitHub, so unfortunately there's not much I can do. I also don't have enough space on any of my other servers at the moment to mirror it.

This is one of the reasons behind PiBakery v2 (which seems to have issues on Windows, so I wouldn't recommend trying that quite yet), as that's a much much smaller download.

davidferguson, 2018-08-20:
Edit: I've just tried downloading it myself, and it worked fine. Maybe just try again?

andrewjw23, 2018-08-20:
Still doesn't work for me

davidferguson, 2018-08-20:
I've just cleared some space on a server and setup a mirror, let me know if this works for you: http://dl3.pibakery.org/PiBakery.Setup.exe

andrewjw23, 2018-08-21:
After a few failed attempts, I finally have the installer!
Now starts the learn curve!
Thanks for your help

kabilramar, 2019-01-13:
David sir when i download PiBakery for Windows 7 after some time it shows Forbidden error. I tried several times whats the reason .

donjuanmon, 2022-12-15:
> Hi David, I have used pibakery before with great success. But now I cant seem to download the windows setup. I tried installing from source but it throws bunch of errors I have to resolve and will take time. Could you please point me to a windows installation please? Many thanks, Kaushalya

This project appears to be abandoned, I'd recommend switching to a raspbian OS cloud-init based approach. You can get the same functionality with a project like https://github.com/jsiebens/rpi-cloud-init