davidferguson/pibakery #38: Node for global variables

Barry4679, 2016-09-18:
PiBakery would be enhanced if there was a top level node for global variables (ie. addition to the existing onfirstboot & oneveryboot nodes).

The graphical interface is attractive, and simple for simple things. But it is fragile. 

Where a xml file is exported, and sent to someone else, it is relatively easy to unhook from the chain, or to screw things up when changing things like network passwords, or directory names buried within RunCommand blocks. ... Also it is a pain to refactor a chain where something like a path needs to be modified. (pasting a substring into an attribute seems to do a replace instead of an insert).  

A global variables chain could be placed at the top, with a block for any element requiring customisation.  ... ?

jsve, 2017-02-02:
I have written a couple of lines to replace stuff inside the Bakery-nodes with strings from a json-file. Check it out here: https://github.com/jsve/pibakery-raztelzig/blob/master/star_replacer.js

Basically you put something like {{{my_long_line_of_code}}} in the node, and it is replaced with whatever is mapped with "my_long_line_of_code" in the json. The replacements are put in a copy of the file. Simple stuff, but might be of some help. All filenames are hardcoded in the script :S

I made this to keep my passwords hidden from the world, while still being able to use github for my bakery-recipies. The same should work for inserting other stuff into the nodes.

davidferguson, 2018-08-31:
This is probably not something I'm going to support.

closed 2018-08-31 by davidferguson