davidferguson/pibakery #40: request to modify boot.txt options

japharl, 2016-09-18:
Would be nice to modify boot options as a new block.  hdmi audio selection, and camera led disable might be two good candidates for this.

davidferguson, 2016-09-20:
That's a good idea - I've already got the IQaudIO block which modifies boot.txt, so I can hopefully reuse some code from that.

Besides HDMI audio and camera led, are there any other popular requests?

japharl, 2016-09-20:
See: http://elinux.org/RPiconfig​ for the full list of options.  So, looks
like you can enable the camera, and do split memory configuration,
overclock, max_usb_current (600ma vs 1200ma)  (and misc. video options
which needs to be done on a case by case basis).  So, prioritized:
camera enable / camera led
overclock
split memory configuration

lurch, 2016-09-23:
...and there's also https://www.raspberrypi.org/documentation/configuration/config-txt.md

But some of these options are probably better done through raspi-config ( https://github.com/davidferguson/pibakery-blocks/issues/8 ) than editing the file directly?

M0JFP, 2016-10-03:
what about the over clocking option I'd suggest a max (safe) of 1000 on pi-3

eddwatts, 2016-11-07:
iv had a look at starting to make a few of these, i needed them myself:
https://github.com/eddwatts/pibakery/tree/master/pibakery-blocks
i have requested a pull to put them for everyone to have access to them

I have working so far:
display mode setting (setdisplay)
Set Audio mode (setaudio)
Set camera led mode (setcameraled)

M0JFP, 2016-11-07:
Hello Team,

Would it be possible to send these additions to any .IMG file.
In our ham radio hobby we tend to create images and share them in the community, with the latest software or drivers for the radio’s etc.
It would be great to be able to edit the image, instead of the vanilla raspbian one.

Thanks for your help and great work on the project so far.

James.

https://github.com/eddwatts/pibakery/tree/master/pibakery-blocks

eddwatts, 2016-11-14:
as far as i know pibakery uses its own version of the img, im sure i have read there looking at making it more comparable with the official img release but its not ready yet - once that's been done it may be able to work with other img files but i dont know