davidferguson/pibakery #41: Have pibakery hosted online

MonkoftheFunk, 2016-09-18:
Perhaps it can be a web service that I can just make recipes on and export to be written later. Maybe could connect to google docs to save to and share from?

davidferguson, 2016-09-18:
I can easily have it able to create and export/import recipes online for editing, but obviously writing isn't possible online. I've thought about this before, and had the idea that the image on the homepage could actually be the web app running, which would achieve what you want.

Funnily enough, the very first version of PiBakery was actually web only, where you would edit a JSON config, submit it to the server, and about 5 minutes later you were emailed a link to download the .img file which had your customisations included. I stopped working on that pretty quickly, as I realised that having to download a 4+GB file every time you wanted a new image would put people off pretty quickly. The technology used was pretty cool, though.

I'll think about adding a web editor/importer/exporter - and I'll report the status back here.

MonkoftheFunk, 2016-09-19:
Thanks ya, even though it wasn't practical from a download, great experience building that service I'm sure! (Rabbit MQ or your own?) I'll keep posted, thanks again for this product and help on previous issue.

niko86, 2017-03-10:
Have you considered having the user download the official raspbian image themselves and your site serving a delta patch of the changes to the official image, obviously a local version of pibakery would need to merge this patch or use some other tool.

Also could you have the program generate an image rather than writing direct to the sd card?

meltwater2000, 2017-06-19:
Excellent work, really glad that this kind of project has gained the support of the community.
Certainly something which is needed and the PiBakery has a very neat interface and very nicely thought out..

I'm interested to see if this can be integrated into Noobs (or PINNs) using the work which @procount had done using noobsconfig.  It should be possible to inject the required files into an image installed via NOOBS and then run all the PiBakery as normal.
I think the only part that will be needed is to have an export of the generated /boot/PiBakery files to drop in after NOOBs has completed.

I'll have an experiment and see if I can make it work, if so I can retire the [PiKitchen](https://github.com/PiHw/Pi-Kitchen) project which was a proof of concept along the same lines.

Ideally, support for injecting the PiBakery into any installation method would make sense (making a common platform for customising installs).
- via NOOBS (via noobsconfig or directly into compressed image files)
- on existing images (via apt package as you've mentioned elsewhere)
- via IMG files (if possible to directly inject into the IMG)

acrogenesis, 2018-01-19:
I'm very interested in having the possibility of having this on a webserver.