davidferguson/pibakery #42: VNC gray screen

Barry4679, 2016-09-20:
I am have trouble with VNC Server (caused by yesterday's update?)
I sign in from my PC but it now just shows a gray screen, but no desktop.
ie. like this https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=129078 

I need to sign into the pi with putty and manually run the tightvncserver command, which responds

	    New 'X' desktop is raspberrypi:2
	    Starting applications specified in /home/pi/.vnc/xstartup
	   Log file is /home/pi/.vnc/raspberrypi:2.log

Then everything comes OK when if I start TightVNCViewer on #.#.#.#:2 

I looked at the vnc log for attempt :1, but see no obvious errors.

I have narrowed my PiB script down to:

### OnFirstBoot: 
* Wifi setup
* install VNC Server
* reboot

### OnEveryBoot: 
* Run VNC Server; default port

It seems that I can access via TightVNCViewer while OnFirstBoot is running, but not after OnEveryBoot.

Something missing from my PiB script? ... I am not sure, but I think this worked ok before yesterday's changes ... not 100% sure because I was also making changes at the time.

davidferguson, 2016-09-20:
I did change the VNC block (I though I was fixing it, although I may have been doing the opposite!).

I think I know what the issue is, but sadly I don't have access to a Pi at the moment to test, but I'll probably roll changes on that block back to before the update, as I know it did work then.

davidferguson, 2016-09-20:
I've found the bug, and I'll try and push out a PiBakery block update for it today.

Thanks for reporting!

davidferguson, 2016-09-20:
I've just pushed out a block update which fixes this. Thanks again for letting me know about this issue.

closed 2016-09-21 by davidferguson