davidferguson/pibakery #44: Custom Scripts Block

denzelwamburu, 2016-09-20:
Great Work,

Can there be a feature for adding a custom scripts, like for example running a nodejs script?

Thanks.

denzelwamburu, 2016-09-22:
@davidferguson not really, but custom commands.

denzelwamburu, 2016-09-22:
Like a shell script for example

davidferguson, 2016-09-25:
Like running a script? There is the `Run command` block, which allows you to input a command to run (such as `touch /boot/test` or `mkdir -p /my/new/directory`) is this what you are meaning?

LowPowerLab, 2016-11-09:
**Run command** should work for what denzel needs, I use it a lot to create all kinds of things
I just wish I could assign a description/title to it.

denzelwamburu, 2016-11-10:
Hello Guys,

I needed to run nodejs scripts after installation and startup.

The new pi Pixel OS comes with Nodejs pre-installed.

Let me give it a try and update you guys.

alandsidel, 2020-11-19:
> 
> 
> Do you mean a block where you can paste in a large block of text, and it would be saved as a file?

That doesn't sound like it's what he meant, but I for one could really use such a block.  I have a ~40 line shell script I need to run on startup and adding 40 individual command blocks is a bit much.  Alternatively, having the ability to copy any local file to a directory in the image would probably be useful in more instances.