davidferguson/pibakery #46: Can't Install on OS X 10.11.4

jamestenglish, 2016-09-21:
On the "Select a Destination" window in the OS X installer I receive the message "PiBakery can't be installed on this disk" when my only hard drive "Macintosh HD" is selected. 

The installer reports I have 32.21 GB available, does the install take up more than that amount of space or am I missing something?

davidferguson, 2016-09-21:
I've had a few reports of this now, and right now I don't know why this is happening. On my Mac (10.10) it installs fine, and I've just tested another Mac (also on 10.10) and that worked fine as well.

Could you go to Window -> Installer Log and paste the contents of all the messages here?

And could you generate a md5 checksum of the installer .pkg and paste that here as well?

jamestenglish, 2016-09-22:
```
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: LSExceptions [0x7fe0a170da90] loaded
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: @(#)PROGRAM:Install  PROJECT:Install-1000
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: @(#)PROGRAM:Installer  PROJECT:Installer-853
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Hardware: MacBookPro11,1 @ 2.40 GHz (x 4), 4096 MB RAM
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Running OS Build: Mac OS X 10.11.4 (15E65)
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: TMPDIR=/var/folders/rb/h1r_j83j6h37dh66l7pn7gxh0000gn/T/
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: __CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: HOME=/Users/jamesenglish
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: SHELL=/bin/bash
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.gPTmUPBgdF/Render
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.wCnrZu6OYj/Listeners
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: PATH=/usr/bin:/bin:/usr/sbin:/sbin
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: LOGNAME=jamesenglish
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: DISPLAY=/private/tmp/com.apple.launchd.lNAyRMcTqp/org.macosforge.xquartz:0
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: XPC_SERVICE_NAME=com.apple.installer.1952
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: USER=jamesenglish
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Env: XPC_FLAGS=0x0
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: PiBakery  Installation Log
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Opened from: /Users/jamesenglish/Downloads/PiBakery.pkg
Sep 21 20:30:25 Jamess-MacBook-Pro Installer[17481]: Product archive /Users/jamesenglish/Downloads/PiBakery.pkg trustLevel=100
Sep 21 20:30:35 Jamess-MacBook-Pro Installer[17481]: LSExceptions [0x7fe0a170da90] unloaded
```

```
$ md5 PiBakery.pkg 
MD5 (PiBakery.pkg) = eed4a625f9151d5e98cee795363f979a
```

davidferguson, 2016-09-22:
I've found the bug in the installer - it turns out that the Installer formatted my install requirements, so when I typed in "5.4 GB" required, it changes it to "54 GB" required. I'm rebuilding the installer now, and will upload it as soon as I can.

Once I've uploaded it I'll add a comment here.

davidferguson, 2016-09-22:
@jamestenglish I've now uploaded the fixed installer, could you try it and see if it works for you?

shawngrimes, 2016-09-22:
new installer worked for me

closed 2016-09-22 by davidferguson