davidferguson/pibakery #47: First Boot - Download files over Open Wifi

lordmuggins, 2016-09-22:
Hi,

I have set up a couple of "First Boot" options. 

1 - Connect to Wifi Network
2 - Download Chromium (wget) from url
3 - Install Chromium - dpkg
4 - Add Chromium, with web page and parameters to autostart file (printf)

Now, when working at home, with WPA password wifi, it works.  When at work, using Ethernet, and open/no password wifi, it does NOT work. Any ideas?

I'm passing this along to someone in the IT department, to create SD cards, so I need it to be as "friendly" as possible.. e.g. the other person just changes the URL in "Step 4" above.

Thoughts appreciated.

lordmuggins, 2016-09-22:
At the very least, my problem is, I need to be able to install packages on first boot. But, the said packages are not in the regular repo..

davidferguson, 2016-09-22:
I'm afraid I won't have access to my Raspberry Pi until the weekend, so I won't be able to test and experiment until then.

In the meantime, what version of PiBakery are you using? (the easiest way to tell is that 0.3.x has support for writing Raspbian Lite as well as Raspbian Full, while 0.2.x can only do Raspbian Full)

If you're using 0.3.x then PiBakery will produce logs on the SD card of the blocks it runs. The firstboot log is located `/boot/PiBakery/firstboot.log`.

davidferguson, 2016-10-02:
Did you manage to have a look in the `firstboot.log` file?

closed 2018-06-29 by davidferguson