davidferguson/pibakery #51: Can't Write to SD Card

jamestenglish, 2016-09-23:
When I try to write my recipe to and SD Card I get:
```
SD Write Fialed
error: EIO: not able to write image -5
Please try writing to the SD card again.
If this error persists, please reinstall PiBakery.
```

I have tried re-writing and re-installing to no avail

davidferguson, 2016-09-23:
Are you trying to write Raspbian Full or Raspbian Lite? What size of SD card are you using? Are you in Windows or Mac? (or Linux?)

AlanHasty, 2016-09-23:
In my case (Win 7, 8GB Sd card) , clicking on the Write button doesn't open anything. So I don't get to choose.  I can see the drive in Windows Explorer. formatted as FAT32.
Any ideas?

davidferguson, 2016-09-23:
@AlanHasty The keyboard shortcut ctrl-shift-i will open the developer inspector. If you then choose Console in the inspector, are there any errors or messages in there?

AlanHasty, 2016-09-23:
Hi David,
I have 5,
Uncaught (in promise) DOMException: The play() request was interrupted by a
call to pause().
Uncaught (in promise) DOMException: The play() request was interrupted by a
call to pause().
Uncaught (in promise) DOMException: The play() request was interrupted by a
call to pause().
Error: ENOENT: no such file or directory, open 'E:\PiBakery\blocks.xml'(…)
Error: ENOENT: no such file or directory, open 'J:\PiBakery\blocks.xml'(…)

It's complaining about pibakery.js : 1057  (J: is the device I'm expecting
to write.)

davidferguson, 2016-09-23:
@AlanHasty Those errors are all perfectly normal. Do you have a USB flash drive that you could test to see if that shows up in PiBakery, to see if it's the SD card it doesn't like?

AlanHasty, 2016-09-23:
Sure, I'll give it a try.

AlanHasty, 2016-09-23:
Same result, Log file attached.

davidferguson, 2016-09-26:
What size of SD card are you using @AlanHasty @jamestenglish? And are you trying to write Raspbian Full or Raspbian Lite?

AlanHasty, 2016-09-26:
I believe I was using an 8G card.

P.s  will this work on Linux? Or just Mac/Win?

davidferguson, 2016-09-26:
@AlanHasty I'm not sure what's causing this error - give me a day or two and I'll have some test code for you to run that should help find the issue.

It will work on Linux (see https://github.com/davidferguson/pibakery/issues/2) and it actually already should run on Linux if you build from source - but I need to do more testing before I package up a .deb and a .rpm for people to use.

dobby, 2016-10-26:
Hi I have the same issue on 2 mac's. This is what the console prints:

[image]

micheld93, 2017-01-06:
Hi, 

it seems I have a similar problem. I posted it there in the meantime, before I found this issue: https://github.com/davidferguson/pibakery-raspbian/issues/27

Unfortunately, I can't provide any console logs because there are no logs. Will Mac OS X delete those logs after reboot? I'm using Mac OS X Mavericks and the newest version of "Pibakery". 

I get exactly the same error message as shown in the first post here. As far as I was able to investigate the problem it seems to be a problem when you do something else in the meantime on the same machine (checking mails for example). But sometimes it don't work properly even if you only have "Pibakery" open. It's strange.  

The SD card is in good conditions (nearly new). Today, after reboot, it has worked perfectly with the very same SD card that made problems in the past.

javi9231, 2017-02-15:
I had the same problem. My system is linux, the sd card of 32gb.
In the log it gave an error, could not find kpartx.
I installed it and it worked without errors.

davidferguson, 2017-02-15:
> I had the same problem. My system is linux, the sd card of 32gb.
> In the log it gave an error, could not find kpartx.
> I installed it and it worked without errors.

`kpartx` is a requirement for the Linux version of PiBakery - but not for Windows or Mac.

@dobby The error message you're getting is coming from the module that I use for writing images in PiBakery. There have been a few updates to this module since I last released PiBakery, so in the next release I'll be updating it (and a few other modules as well).

I have occasionally received that error when writing SD cards, but whenever it occured for me, it was always because I had accidentally set the write-protect switch on the SD card. Sometimes I even found that the action of inserting it multiple times caused the switch to be set to protect. If you're experiencing this error - that's the first thing you should check!

micheld93, 2017-02-15:
This was exactly what I thought but unfortunately, microSD don't have any protection switch. So, it may be a problem but not in my case...I use a microSD writer so I don't need the converter microSD-to-SD adapter.

RogueM, 2017-04-12:
> kpartx is a requirement for the Linux version of PiBakery

can the front page README be updated to reflect this please. In my case though I can't seem to access the dev log - does this work on linux?

adamkruzics, 2017-06-13:
hi guys–

same problem here too, couldn't use my old PiBakery – as it required update; downloaded and reinstalled ( x3 during the day ... :-) ), bought 2 new micro SDs, spent like 3 hours on it today but nothing changed, same error msg all the time.
was here any update on this matter? shall I change my Mac for an older Win? would that help?

appreciate any work-around idea!

jherrin6, 2019-11-11:
I ran into the same issue on mac os x catalina.  I was able to work around the issue by opening PiBakery from the command line with sudo:

`cd /Applications/PiBakery.app/Contents/MacOS/`
`sudo ./PiBakery`

andrewgy8, 2019-12-19:
I also got this error running MacOS Catalina 10.15.1. And the above solution worked like a charm.

micheld93, 2019-12-21:
Yes, this is working for me as well.

lwerner-lshigh, 2019-12-27:
Same issue, also with Catalina, solution works! Thanks!

rafaelvannucci, 2020-01-30:
> I ran into the same issue on mac os x catalina. I was able to work around the issue by opening PiBakery from the command line with sudo:
> 
> `cd /Applications/PiBakery.app/Contents/MacOS/`
> `sudo ./PiBakery`

Works well

Solilord, 2021-01-18:
> I ran into the same issue on mac os x catalina. I was able to work around the issue by opening PiBakery from the command line with sudo:
> 
> `cd /Applications/PiBakery.app/Contents/MacOS/`
> `sudo ./PiBakery`

This works as well on BigSur.! Thanks!

ben-nn20, 2021-02-16:
Worked for me too on Big Sur. Thanks!