davidferguson/pibakery #56: Ability to add multiple block sources

dsbert, 2016-09-25:
It would be nice if you could add your own git repos for blocks in addition to the official source.

davidferguson, 2016-10-02:
That's an interesting idea, and certainly one I'll investigate. I can see this being useful for schools or clubs that would like custom blocks for them, but wouldn't want those blocks available for everyone who uses PiBakery.

I can probably have it so you would just fork the pibakery-blocks repo, and then edit a setting to point to your fork rather than the main one.

meltwater2000, 2017-06-19:
This would be very useful for development and testing too.

davidferguson, 2018-06-29:
This is now a feature in PiBakery v2 - although currently it isn't exposed in the UI. It'll be added in the future.