davidferguson/pibakery #57: give more info on the "processing next boot script" screen

Godley, 2016-09-25:
Right now my pi is stuck at 50% on the run command block. More info on which block I'm on and maybe some debug output would give more assurance it's not got stuck or something

davidferguson, 2016-09-26:
If you're using the latest version of PiBakery (the easiest way to tell is that the latest version has support for writing Raspbian Lite as well as Raspbian Full, while older versions can only do Raspbian Full) PiBakery produces a log file located `/boot/PiBakery/firstboot.log`.

Have a look in there to see what went wrong, and in a future version of PiBakery I will try to add some debug output onto the progress bar screen.

Barry4679, 2016-09-26:
Or if you are on an earlier version, and don't want to repeat the 1gb+
download just yet, you could log to a file with Run command blocks:
    echo "synaptic installed" > ~/PiBakeryInstall.log
followed by a series of
    echo "cherrypy installed" >> ~/PiBakeryInstall.log