davidferguson/pibakery #58: way to display IP address when GUI is booted?

Godley, 2016-09-25:
I have my device working with the official touch screen. If I run "ipconfig" on every boot I'm guessing it's going to just run that before the GUI boots. Is there a way or could there be a way to have "On GUI loaded" or something scripts?

davidferguson, 2016-09-26:
There's not a way to have a "On GUI loaded" block yet - although it's a good idea and one I'll consider for future versions of PiBakery.

What I have seen other Pi users do (and thought it was a brilliant idea) is have the IP address edited into the desktop picture, so when you boot into the GUI it will show you what the IP address is straight off the desktop. In the next block update, I'll try and add in a block that adds support for this.

Godley, 2016-09-28:
I would be interested in helping you clear up some of these btw, but on the RPi side I have no idea where you'd even start - might be an idea to start writing some developer docs on more in depth stuff like this?

AlexSwensen, 2016-10-10:
A simple Python script could do this!

Godley, 2016-10-12:
An even simpler bash script would also do this, but my point is the python script would (I predict) run before you boot to the GUI. So you'd have to be very quick to spot it. Unless there's some way already in raspbian to have a python/bash script run after the GUI is loaded.

Barry4679, 2016-10-13:
You can see the IP address if you hover over the the WiFi icon on the RPi task bar. Does this help? I don't have the touch screen.

Godley, 2016-10-13:
that *might* do the trick if touch can make the cursor hover rather than open the wifi tab. I'll try that out, cheers.