davidferguson/pibakery #62: Alternative destination for the Raspbian Images

berryelectronics, 2016-10-03:
Add an option to put the raspbian images into a different folder (like an second hard drive) and seperate the pibakery gui and the images, because they really consume much hard drive space and on something like almost every >200-300$ laptop where you only have 32/64gb internal storage but an expandable SD card, having an 1-2gb iso on the main storage is a killer

CleoQc, 2016-10-11:
ITA... Allow for the image to be independent of PiBakery. This would also allow people to have access to other images, as long as they can get PiBakery to point to them.

carlosperate, 2016-10-11:
@CleoQc when you say "other images" do you mean other OS, like what's discussed in #31? Or other versions of Raspbian?
Taking in consideration some of the blocks (or future blocks) might be specific to newer functionality/software added to Raspbian, having to also track the version of the OS that the user selects for pibakery (as it also need to edit the OS image) sounds like a can of worms. I think that having to use the latest Raspbian that's provided by pibakery is not a bad trade-off taking in consideration the general target audience for the tool.

CleoQc, 2016-10-12:
It is indeed related to [#31](https://github.com/davidferguson/pibakery/issues/31) but what I have in mind is more like a pre-made version of Raspbian. So it's still Raspbian, but already fully loaded with many scripts, libraries, and special tweakings. 
http://www.dexterindustries.com/raspberry-pi-robot-software/
We already publish our own image, it's a tweaked Raspbian. We'd love to have PiBakery to let users change hostname and set up wifi.

carlosperate, 2016-10-12:
You can check out https://github.com/davidferguson/pibakery-raspbian to see what changes are needed, but I still maintain that being able to support multiple versions of Raspbian (or other OS) might end up making pibakery more complex than it needs to be. The app will have to be able to identify OS version, and blocks will start targeting specific versions or OS as well, some blocks will have to have different outputs depending on versions, and there will always be combinations or odd corner cases not covered correctly, so maintenance will get exponentially more difficult.

davidferguson, 2018-06-29:
No longer relevant for PiBakery v2

closed 2018-06-29 by davidferguson