davidferguson/pibakery #63: PiB v0.3.2 cannot open recipe from earlier version

Barry4679, 2016-10-04:
Version 0.3.2 cannot open a recipe from earlier versions of PiB if the recipe contains an "Install VNC Server block"

No error message, but the RPi work space remains empty. If I remove the install vnc block, with an older version of PiB, the recipe opens ok.

The ctrl+shift+i inspector shows
<<<
undefined:1 Uncaught (in promise) DOMException: The play() request was interrupted by a call to pause().
undefined:1 Uncaught (in promise) DOMException: The play() request was interrupted by a call to pause().
undefined:1 Uncaught (in promise) DOMException: The play() request was interrupted by a call to pause().
pibakery.js:955 Error: ENOENT: no such file or directory, open 'D:\PiBakery\blocks.xml'(…)(anonymous function) @ pibakery.js:955
blockly.js:1006 Ignoring non-existent field 1 in block vncstart
blockly.js:1008 Uncaught Next statement does not exist.Blockly.Xml.domToBlockHeadless_ @ blockly.js:1008Blockly.Xml.domToBlockHeadless_ @ blockly.js:1008Blockly.Xml.domToBlock @ blockly.js:1003Blockly.Xml.domToWorkspace @ blockly.js:1001(anonymous function) @ pibakery.js:2346(anonymous function) @ D:\Program Files (x86)\PiBakery2\resources\app\node_modules\fs-extra\node_modules\graceful-fs\grace…:78tryToString @ fs.js:455readFileAfterClose @ fs.js:442
>>>

dsbert, 2016-10-13:
This is related to #55

davidferguson, 2018-06-29:
This was because I changed a few things in different releases of the blocks, and therefore PiBakery didn't know how to read older blocks with different fields/etc.

PiBakery v2 *should* open a recipe exported from *any* version of PiBakery - unless it includes a custom block, but that issue will be discussed under #55.

closed 2018-06-29 by davidferguson