davidferguson/pibakery #64: No-OS installer download

avdmatamoros, 2016-10-04:
Would it be possible to have the installer/repo download without the OS extra weight and handle it as a post-install update/download? I got network download issues near the end of the 1.3gb download a couple of times and all I wanted was the app installer to use a custom image.

haiqu, 2016-10-26:
Since the bulk of the file is the RasPi OS this wouldn't really help.

davidferguson, 2016-11-02:
Yes, currently PiBakery requires a modified version of Raspbian to work, so this wouldn't solve anything as you'd still need to download the raspbian-pibakery.img and then integrate it with PiBakery afterwords.

However, in a future update I plan to support non-modified Raspbian images, which will mean that PiBakert by default won't come with the image bundled, and you'll be able to drag-drop any Raspberry Pi image into PiBakery.

Gabbalo, 2016-12-08:
It would be nice, to not only support non-modified Rasbian images but modified ones, too, such as cozy.io. 
If they are based on Raspian, it should not be complicated to include at least most blocks (wifi mostly appreciated :) ) with those images, should it?

doej1367, 2017-01-28:
This would definitely help! I always get the images from the Magazin-DVDs by CHIP or MagPi. So I dont have to download them. (this would take days anyway, using my 64 kbit/s-Internet...)

doej1367, 2017-01-28:
Would it be posible to make a "PiChangery"? A version that onely supports the "Editing an SD Card"-Feature, that has only this 12MB size? Or does this also need the OS-image?

davidferguson, 2017-01-28:
The next major release of PiBakery will actually have this feature - you'll be able to download PiBakery without an image, and then use any debian based image with PiBakery. I'd hoped to get it out in 2016, but schoolwork and exams have delayed things a bit, so I'm now looking for a February release for this (and a few other new features as well!).

davidferguson, 2018-06-29:
> The next major release of PiBakery will actually have this feature

Now been released - sincere apologies for the unacceptably long delay.

closed 2018-06-29 by davidferguson