davidferguson/pibakery #65: Cannot no longer remote into the RPi after lastest RealVNC patch.

Barry4679, 2016-10-05:
I installed Pixel onto my RPi using PiBakery. I was able to log onto the RPi from my PC using VNC Viewer (thanks :))

The latest sudo apt-get dist-upgrade now includes an update to realvnc-vnc-server. I installed the upgrade, but could then no longer remote into the RPi.

What is setup by PiBakery's "WNC Server at BOOT" block? 
Does this need to run on EveryBoot? I had it just on OnFirstBoot, and this was working on successive boots prior to the upgrade.

What do I need to do to re-establish VNC the way the PiBakery had it working?

I tried running "vncserver". This did enable me to remote in if I switched from #.#.#.# ti #.#.#.#:1, but it then ignored the changes that I had made to my /boot/config.txt file to configure Pixels' resolution to suit my PC display.

Barry4679, 2016-10-06:
Running "vncserver" was the wrong thing to do. 
I needed to SSH in, and then sudo raspi-config. , and then Advanced Option, to re-enable VNC Server.

Would having a "WNC Server at BOOT" block in EveryBoot have avoided this?

Barry4679, 2016-10-06:
Conclusions; with a headless RPi:
* having a "VNC Server at BOOT" block in EveryBoot would have avoided this this situation
* or use VNC in virtual desktop mode; ie. don't have a "VNC Server at BOOT" anywhere, but run a command block in EveryBoot with "vncserver -geometry XRESxYRES" .... eg. vncserver -geometry 1024x768

as I was advised in this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=161341

AlexSwensen, 2016-10-10:
If your issue is solved, it would be best to close it.  Thanks for the helpful info!

Barry4679, 2016-10-10:
Thanks. I am not familiar with the Github rights set up. I hadn't noticed that I could do that.

closed 2016-10-10 by Barry4679

AlexSwensen, 2016-10-11:
Both you and those that are maintainers of the repo can do that. I, for example, cannot. I'm just trying to be a helpful passer-by! 👍

davidferguson, 2016-10-11:
I'm considering changing the VNC block back to just a **Run VNC server** block which you'd have to put under **On Every Boot** as the new one seems to be causing a little bit of confusion over where it should be placed.

I could also add into the block options such as the resolution, port, etc. Would this be well received, or just overcomplicate things?

Barry4679, 2016-10-11:
What is the PiBakery stance re people running pre-Pixel Rasbian? Wouldn't they need the old blocks? See my issue here https://github.com/davidferguson/pibakery/issues/63

I think that a setting for resolution would be a good thing. That way we could send someone a PIB xml file. They open it, set their WiFi pwd and screen resolution, and they are good to go.

AlexSwensen, 2016-10-12:
@davidferguson i too find myself confused by the default block, and resorted to just manually setting up my realvnc-server and just using PiBakery for enabling wifi config, changing password, and regenerating ssh-keys. 

I would leave it as a "first boot/next boot" block, but allow the user to specify it permanently enabled to auto-start vs a 1 time session. Maybe thats overly complicated. I do second the resolution and port, but for headless instances i see resolution being a painful change unless the `/boot/config.txt` file is modified. (only applies for x11 vnc instances)

Barry4679, 2016-10-12:
<<<What is the PiBakery stance re people running pre-Pixel Rasbian? Wouldn't they need the old blocks? See my issue here #63>>>

I see that I haven't explained this very well. Just in case it is unclear:
1. I was running the pre-Pixel version of PiBakery
2. I developed a working PiB recipe
3. I then took the PiB upgrade, which included Pixel (but also had removed some blocks I had used?)
4. I was unable to open my recipe with the new version of PiB (no error message, it just didn't open) ... so it was either reinstall the previous version of PiB, and then go back and forth between each, making deletions until I found a recipe which could open with the new version of PiB ... or try making manual cuts to the xml