davidferguson/pibakery #68: Can't edit an existing setup

sachinbhatnagar, 2016-10-20:
I burnt a setup of blocks on the card and have been using it on the Pi for the last couple of days. Now I wanted to update the configuration, so I tried the instructions given in the docs but PiBakery just doesn't give me an option to load existing configuration from the card. The card is there, there are no blocks in the workspace. Nothing happens. I'm on MacOS Sierra.

Barry4679, 2016-10-20:
Maybe you are seeing this?
https://github.com/davidferguson/pibakery/issues/63

sachinbhatnagar, 2016-10-21:
Nopes. Just that when i plug in the SD card on my mac, piBakery should give me an option to load up the configuration on the card. This it doesn't. And so i'm unable to edit my setup.

haiqu, 2016-10-27:
This can happen in two circumstances under Windows:

1. Card is assigned a different drive number by the OS.
2. Card is inserted before running PiBakery.

I can't comment on whether either of these is an issue on a Mac.

RichardBronosky, 2016-11-24:
I'm having the exact same experience. I strongly suggest you try a little less magic with this application. Allow the user to do some things explicitly rather than them supposed to just happen automagically.

rhuairahrighairidh, 2017-01-05:
I'm also having the exact same problem.
I have macOS Sierra as well.
I've tried editing a SD card that has been booted in a pi and one that has just been written by pi bakery. Neither triggers the "edit existing card" dialog box .

rhuairahrighairidh, 2017-01-06:
I've been digging around in the code to try and find the problem.
The problem seems to be happening in the module drivelist-scanner.

It sets up a loop that polls the module drivelist every second to see if the list of available drives has changed.
During this it removes from the list any drive that is listed as a 'system' drive by drivelist.
This should work fine as the SD card reader is normally not a 'system' drive.

However for some reason drivelist is returning the list of drives incorrectly. The SD card is listed as a 'system' drive and all the other drives are missing their size and mount point.
So drivelist-scanner strips out the SD card reader because it is now a 'system' drive and pibakery thinks that no new drives have been added.

I don't know why the list of drives is being returned incorrectly. Pasting the function into the console and running it gives the correct results.

Also I don't really know javascript so its possible I'm missing something.

RichardBronosky, 2017-01-06:
Great research, rhuairahrighairidh. I gave up after I saw what nest of spaghetti the code was.

davidferguson, 2018-06-29:
> I strongly suggest you try a little less magic with this application. Allow the user to do some things explicitly rather than them supposed to just happen automagically.

This was a good idea, and has been implemented in PiBakery v2.

With the way PiBakery v2 handles editing of SD cards, this should no longer be an issue. Instead of everything happening automatically when an SD card is detected, the user now has to select the SD card from a dropdown list.

closed 2018-06-29 by davidferguson