davidferguson/pibakery #71: PiBakery needs root to be run

maweki, 2016-10-28:
I am running Fedora 24 and I tried to follow the instructions but it fails to launch with an error dialog saying "Please run this application as root or administrator".

Pibakery should use other mechanisms to elevate its privileges for two reasons.

One: If you have a local node installation via `nvm` which is quite common (since `node` packages of distributions are usally quite old), using `sudo` doesn't work without augmenting `PATH` information

Two: Once wayland is some form of standard on the desktop, graphical applications called via `sudo` (or any user that's not the one running the Wayland instance) will not have access to the graphics server. Running as root will then be impossible, if you don't want to run a whole privileged graphical session.

davidferguson, 2018-06-29:
The main app no longer requires root to run, however writing does require this. The method of elevation in PiBakery v2 is almost identical to older version of PiBakery, as it's the only way that I currently know how to do.

Comments and feedback on other methods of elevation would be appreciated.

gauntface, 2018-09-28:
Why does writing require sudo? I would have assumed a user executing the script would be able to write to the user's home directory.

Picard0147, 2018-10-20:
> Why does writing require sudo? I would have assumed a user executing the script would be able to write to the user's home directory.

Writing mentioned here is about writing modified raspbian image on SD card so it's not clasic writing file but "dd if=xxx of=yyy" where xxx is image on your drive and yyy is sdcard (/dev/sd? or /dev/mmcblk?).