davidferguson/pibakery #75: npm install fails with error

metaviews, 2016-11-02:
Hello! Trying to install pibakery on my Ubuntu 16.04.1 laptop.
Installed latest version of nodejs. I'm getting these errors when trying to run npm install:

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: elevator@^1.0.0 (node_modules/elevator):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for elevator@1.0.0: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: removedrive@^1.1.1 (node_modules/removedrive):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for removedrive@1.1.1: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: diskpart@^1.0.0 (node_modules/resin-image-write/node_modules/diskpart):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for diskpart@1.0.1: wanted {"os":"win32","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: macmount@^1.0.0 (node_modules/denymount/node_modules/macmount):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for macmount@1.0.0: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN PiBakery@0.3.2 No repository field.

Let me know what I may be missing

Godley, 2016-11-05:
Those all look like warnings for other platforms, so nothing to worry about...did the command complete/have you tried running npm start?

metaviews, 2016-11-07:
Ha! Indeed, running npm start worked just fine. The warning errors threw me off. :)

closed 2016-11-07 by metaviews