davidferguson/pibakery #81: Cannot locate "the blocks"

RichardBronosky, 2016-11-24:
I feel like a complete idiot, but I cannot find the blocks in the pi-bakery app. I wanted to try https://github.com/davidferguson/pibakery-blocks/tree/master/retropiesetup so I cloned that repo and dragged the `retropiesetup` folder to the pibakery window and it says I'm about to overwrite an existing block. What? I can't find the block. I can find any of the block from the pibakery-blocks repo or https://github.com/davidferguson/pibakery/tree/master/pibakery-blocks

What am I missing?
[image]

tenten8401, 2016-11-25:
PiBakery-blocks are all installed by default, he just made a separate repo for them. I'm unsure why something like RetroPie is disabled but I'm willing to bet it's still a work-in-progress

RichardBronosky, 2016-11-25:
If we compare `category:software` in these 3 locations, you'd expect it to show up in the app.
https://github.com/davidferguson/pibakery-blocks/blob/master/retropiesetup/retropiesetup.json
https://github.com/davidferguson/pibakery/blob/cb9b3e5f9b2999177c212fa34adfc7e82a6d55e5/pibakery-blocks/categories.json
https://github.com/davidferguson/pibakery/blob/cb9b3e5f9b2999177c212fa34adfc7e82a6d55e5/app/index.html

davidferguson, 2016-11-25:
> I'm unsure why something like RetroPie is disabled but I'm willing to bet it's still a work-in-progress

Yes, if a block hasn't been enabled yet, it's probably because I found an issue with it. From memory, I think Retropie is a interactive install, which obviously doesn't work with PiBakery.

RichardBronosky, 2016-11-26:
How do scripts get enabled and disabled?

Also, how do you do first run scripts on Raspbian via Mac? I tried to make sense of your code but kept getting lost. I really don't know what you could do to an SD card to cause Linux to run a script at boot.

tenten8401, 2016-11-26:
@RichardBronosky 
[image]

RichardBronosky, 2016-11-26:
That is what I meant, but you can't mount that partition on a Mac. Only the /boot partition can be mounted.

tenten8401, 2016-11-26:
@RichardBronosky http://osxdaily.com/2014/03/20/mount-ext-linux-file-system-mac/

RichardBronosky, 2016-11-29:
I've been down the FUSE rabbit hole. It was disaster. Is that how pibakery is doing it?

JasperAlgra, 2016-12-01:
I'm running into the same issue as @RichardBronosky . I tried to add a custom block, but it won't show up in the GUI.

Block:
```
{
	"name": "narrowcasting",
	"text": "Narrowcasting \\nWebsite: %1",
	"script": "narrowcasting.sh",
    "args": [
        {
            "type": "text",
            "default": "http://theslideshow.net/#simple/raspberry%2Bpi",
            "maxLength": 0
        }
    ],
	"network": false,
	"continue": true,
	"type": "setting",
	"category":"setting",
	"supportedOperatingSystems": [
		"raspbian-pibakery.img",
		"raspbian-lite-pibakery.img"
	],
	"shortDescription":"Run website for narrowcasting",
	"longDescription":"Run website in Chromium Browser Kiosk mode"
}
```
I've based this json on the authorizekey block from your repo.

Also, in windows dragging and dropping blocks doesn't work, probably due to the elevation of the bakery executable, so I'm adding the block by 'ctrl shift plus'. After adding it states `imported block. 'narrowcasting' will be available to use until you next restart PiBakery.` but after a restart it doesn't show in any of the categories.

Is there any was I can debug this or help?

JasperAlgra, 2016-12-01:
At the moment I see some pretty random behaviour. Almost immediately after posting my last message the narrowcasting plugin/block I've made was indeed available in the setting list. I even succeeded to make a working image out of it. Export/import of the xml of buildingblocks also worked.
but.. after re-importing the custom block import of XML silently fails + the block is not available anymore.

Some more verbose messaging would be great. Can I enable that somehow? Or can I run the program via npm on windows to debug?

Cheers,

Jasper

javi9231, 2017-02-14:
I have managed to show the modules.

The problem is that they are not written in info.json from the directory pibakery-blocks
[image]
You have to write the modules you need, 
When restarting pibakery you can see them
[image]

Regards