davidferguson/pibakery #87: Still Active?

yombo, 2017-01-14:
Just wondering if this still active? It's been several months with no activity.

Would really appreciate a new release with updated Jessie images for Windows and Mac (and I saw a deb package was on the todo list).

I found this:
https://github.com/dalmago/pibakery-raspbian

I'm able to use these images locally/manually, but unsure how to create a package for MacOS and an installer for Windows.

yombo, 2017-01-14:
Does anyone know how make an installer for Windows or Mac for this? Point me to some websites and I'll tackle this. Easy enoilugh to fork and create a release, just dont know how to create the installers.

davidferguson, 2017-01-15:
> Just wondering if this still active? It's been several months with no activity.

Really sorry for the lack of activity from me these last few months - I've just been really busy with schoolwork, university applications and exams, all of which take priority over PiBakery.

I'll have some time in February, where I'll look at all the pull requests and issues, and I'll also upload an update with some new features I've been working on too.

yombo, 2017-01-16:
David,
  No worries. I've managed to create an updated build with blocks from other forks. I'm also able to package these up for Windows and MacOS, but holding off due to one last thing:

  I've taken the latest released Raspbian Jessie images from a few days ago. I wrote a little script that pulls from the pibakery-raspbian repo and places the files on the new image.

  However, this only seems to work on the PIXEL image, not the shell image.  On the shell image, in the /var/log/ folder, a grep for "PiBakery" shows that the scripts started. However, nothing in /boot/PiBakery/  - no firstboot.log. The blocks don't run.

  Did you have to do anything different for the shell image versus the PIXEL image?

Thanks.

davidferguson, 2017-01-17:
> However, this only seems to work on the PIXEL image, not the shell image. On the shell image, in the /var/log/ folder, a grep for "PiBakery" shows that the scripts started. However, nothing in /boot/PiBakery/ - no firstboot.log. The blocks don't run.

I don't have access to my files at the moment, but from memory there seems to be a slight bug in the Raspbian Lite image, where the command `systemctl get-default` returns `graphical.target` rather than the expected `multi-user.target`. The file [console.sh](https://github.com/davidferguson/pibakery-raspbian/blob/master/opt/PiBakery/console.sh) checks this, and it fails.

You should be able to get round this by simply removing the check.

yombo, 2017-01-20:
Thanks for the pointer. I've removed the check in the console.sh file for shell only (lite) images.

closed 2017-01-20 by yombo