davidferguson/pibakery #95: vnc server block

nunziofiore, 2017-02-17:
Hi 
 I read on pibakery web site that there is a block INSTALL VNC SERVER and a RUN VNC SERVER AT PORT xxxx
But I have only that...
[image]
how can I have also the other?

nunziofiore, 2017-02-20:
to manage that problem I wrote that :

[image]

if you run from a terminal the command I listed in pibakery, you can set the password for vnc by script and in next boot run vncserver, in this way the end user can use vnc client without write ssh or something he doesn't know.

BUT: when I make this instruction in pibakery the password is not setted.
It's possible there' s a bug when you use a ">" that pibakery change in &gt; for xml purpose? How I can avoid this error?

Barry4679, 2017-02-20:
Hi

<<< I read on pibakery web site that there is a block INSTALL VNC SERVER and a RUN VNC SERVER AT PORT >>>

I think that the web site may be out of date. The latest version of Raspbian included a pre-install version of RealVNC
https://www.raspberrypi.org/blog/introducing-pixel/

I think that was a change to Pi Bakery following that. It is touched one here
https://github.com/davidferguson/pibakery/issues/65

When trying to work through related issues I found the following thread useful
https://www.raspberrypi.org/forums/viewtopic.php?f=66&t=161341

nunziofiore, 2017-02-20:
Thankyou for your answer @Barry4679 . Do you know why the command

`echo password | vncpasswd -f > $HOME/.vnx/passwd`

if launched from command line it runs but if I put it as in the picture in a RUN COMMAND block of Pibakery, it seems that is not launched from pibakery? 
It seems that run command is bugged in some way.

nunziofiore, 2017-02-20:
I tried several RUN COMMAND but no one seems to work. I tried also a simple "touch file" but nothing. I think that the commandrun block is bugged

Barry4679, 2017-02-20:
I am not experienced with Linux Nunzio, and I don't have a Raspberry Pi
here at the moment.

Look for the PiBakery folder inside the Rpi. There is a run log file there.
It may hep you.

Barry4679, 2017-02-20:
I made heavy use of the Run command. I had no issues with it.
Maybe you are having authority issues? Try changing User to Root.

nunziofiore, 2017-02-20:
tried both as pi and root but nothing... now I'll retry and I'll read the log that I didn't know that there was one. I'll write something tp you in few hours

nunziofiore, 2017-02-20:
Ok I've got the right recipe to avoid the problem. Thank you 
[image]

closed 2017-02-20 by nunziofiore