davidferguson/pibakery #96: Stuck on wifisetup

mikecarr, 2017-03-27:
The UI is stuck on wifisetup, not sure if this is a Pi Zero W thing or not.

mikecarr, 2017-03-27:
Apparently it does go through but takes a really long time, are there any logs ?

davidferguson, 2017-03-27:
The `WiFi Setup` block can actually take a while to process, because it will wait up to 5 minutes trying to establish a network connection (after 5 minutes it will continue regardless).

This is because many people use a block that requires a network connection (such as `Install Software`, or `Download File`) straight after the `WiFi Setup` block, expecting the connection to be established.

Therefore to accomodate this, the WiFi setup block will enter the details into the `wpa_supplicant.conf` file, and then repeatedly try and ping google to see if a connection has been established, before continuing (and if 5 minutes passes and a connection hasn't been established, gives up and continues anyway).

Hope this shines a little light on the situation.

(And if you do want to check the logs, have a look at `/boot/PiBakery/firstboot.log` and `/boot/PiBakery/everyboot.log`)

closed 2017-03-31 by davidferguson