davidferguson/pibakery #97: Run pibakery from the command line

eskatos, 2017-05-01:
It would be awesome to be able to run `pibakery` from the command line to load an exported setup file and use it to create a sd card in a single step.

eskatos, 2017-05-01:
This could ideally work without showing any GUI. But it would already be very nice to be able to automate the creation of SD cards from exported setup files even if a GUI shows up.

dannystaple, 2017-05-06:
I suppose it would mean taking the node core (not the electron bit) and creating a nodejs command line wrapper around it - something like

    node bakemeapi /foo/bar/recipe.xml /mnt/sdcard

Which would read the recipe foo/bar/recipe.xml, and write it to the device /mnt/sdcard. 
This does sound like a great idea.

Aboisier, 2017-05-13:
Hi,

I would be interested in implementing this feature. 

I started looking into it, and realized that, to make it in a clean way, it would require compartmentalizing a few methods in their own script/package (aka WriteImage and the related methods). It would be some kind of device/drive/sd helper.

After, it would be a matter of adding the command to the package, which would be pretty simple. 

Is that something that would be interesting if I do a PR?

Hitabis, 2017-06-30:
Posted a pull request for this issue. 
https://github.com/davidferguson/pibakery/pull/106

AlexandruBuhai, 2019-10-15:
+1 to have this implemented in order to automatize pibakery