davidferguson/pibakery #98: OS Update Failed on macos sierra

miguelchico, 2017-05-02:
Hi.

I'm getting this error every time I open PiBakery:
At the beginning a message appears saying a new Raspbian lite is available.
[image]

After I try to update an error message appears:
[image]

benjamin-hering, 2017-05-20:
I got a similar error in macOS Sierra (10.12.4). Tried reinstalling PiBakery 0.3.2 with no success. 

Eventually I tried downloading the latest update manually. I discovered where the Base OS files were stored on my local machine `/Library/Application Support/PiBakery/os` and downloaded the [latest pibakery specific raspbian lite version](https://github.com/davidferguson/pibakery-raspbian/releases) from there and extracted it into the OS directory and named it the same as the last Raspian-lite image. Still got the error, though, even though with keeping the same naming convention.

davidferguson, 2017-06-25:
PiBakery should will display the update prompt when the version number in the file `/Library/Application Support/PiBakery/os/images.json` is less than the version number in `https://github.com/davidferguson/pibakery-raspbian/blob/master/images.json`

The actual error is where it's failing to download the image. If you use the shortcut CMD-Shift-I PiBakery will open the `Developer Inspector`, where if you go to the `Console` tab, you'll see more detailed error details. If this error is still occurring, could you post the detailed errors from that window here?

miguelchico, 2017-06-25:
Hi!
This is not happening anymore.
Thank you!

closed 2017-06-25 by miguelchico