davidferguson/pibakery #99: Torrent download as an option

Fedor-Lyanguzov, 2017-05-02:
Hi there!

Downloading directly from site is pain in all body parts here in Russia. Maybe you could add torrent download?

Thanks!

zeropage-io, 2017-08-02:
Same for Germany. No chance to download. An alternative way to download would be appreciated!

davidferguson, 2017-08-02:
The problem with a torrent option is that it requires me to have at least one server seeding it at all times. Hosting it on GitHub (as I do now) means that I don't have to pay for the bandwidth.

I'll have a look at a torrent option, as this has been a heavily requested feature, but it won't happen for a two weeks at least, as I'm away at the moment.

For now, I've temporarily mirrored the files onto a server I have in France, which should hopefully allow you to at least download it. http://pibakery.org/downloads/

Fedor-Lyanguzov, 2017-08-02:
> The problem with a torrent option is that it requires me to have at least one server seeding it at all times. 
I don't think so. You could seed from your home computer for example, for about 1 month. This could be sufficient for new all-time seeders to emerge. This could work if release cycle is longer than 1 month.

davidferguson, 2017-08-02:
Sadly the release cycle of Raspbian is generally about a month (sometimes longer, sometimes shorter), which makes it difficult. I may try it, and see how it goes. I've looked at the torrent spec, and I *believe* there may be a way to update the content of torrents, which I'll have to investigate.

I'll reply here with the results.

joshghent, 2017-12-03:
@davidferguson As far as I can see, you can download raspbian via torrents. Perhaps if you distributed *just* Pibakery as a torrent and then had the option to import raspbian then people could download the most recent version of Raspbian if they wanted. This would also cut down on the size of the torrent for you.
I have a spare server from Digitalocean that I don't mind using as a torrent box for this. And happy to contribute code wise to make this a feature!

davidferguson, 2018-08-31:
PiBakery v2 (beta) no longer bundles the raspbian images with the program, making it much much smaller, and a torrent option no longer necessary.

Sorry I never added this in for v1, I did experiment with it for a bit but never quite got the whole torrent-updating thing working.

closed 2018-08-31 by davidferguson