davidferguson/pibakery #100: Windows 10 can't write SD card

ShonkyCH, 2017-05-10:
Windows 10 64 bit version 1607

Interface loads and runs. Requests elevated privileges no problem (but same problem occurs if Run As Administrator)

MicroSD card is showing (was a NOOBS card but formatted to blank FAT32) as D:\

PiBakery detects card when attached based on "Error: ENOENT: no such file or directory, open 'D:\PiBakery\blocks.xml'" console messages as expected since card is blank.

When writing, main interface shows nothing. Viewing console errors with:
```pibakery.js:1796 Uncaught TypeError: Cannot read property 'path' of undefined
(anonymous function) @ pibakery.js:1796
scripts.run @ C:\Program Files (x86)\PiBakery\resources\app\node_modules\drivelist\lib\drivelist.js:61
childProcess.execFile @ C:\Program Files (x86)\PiBakery\resources\app\node_modules\drivelist\lib\scripts.js:104
exithandler @ child_process.js:205
emitTwo @ events.js:106
emit @ events.js:191
maybeClose @ internal/child_process.js:852
ChildProcess._handle.onexit @ internal/child_process.js:215
```

blitzter, 2017-05-31:
This is the error that I see after the one you have posted.
Uncaught TypeError: Cannot read property 'path' of undefined 

Looking at the code, it looks like it doesn't account for drives which do not have a mount point.
If before the condition on this line 1796, we can also put a condition that path should be present, this issue should be solved

davidferguson, 2017-06-25:
As said in #105, this should be fixed in the latest update. Could you please uninstall your current version and download and reinstall the newest to see if that fixes it?

imgrant, 2017-08-03:
I have this problem too, but not sure if it is the same one referenced in #105, as I am running the latest (today) version (v1.3.2).

The only error I have in the console is the "pibakery.js:959 Error: ENOENT: no such file or directory, open 'F:\PiBakery\blocks.xml'(…)" when it detects the SD card.
When I try to write, after "Start Write", I confirm Yes and then the error message (in the interface, nothing in the console) reads:

"SD Write Failed

Error: Error whilst writing

Please try writing again, or reinstalling PiBakery

If this error persists, please reinstall PiBakery."

Needless to say, I tried writing again and reinstalling. No dice. How else can I debug?

imgrant, 2017-08-04:
Turns out my problem was a read only SD card (caused by a faulty card reader, I think) -- the card/device showed up as read only in Disk Management, and Win32DiskImager gave an error (1117) when trying to write. I guess PiBakery was not getting the read only error, so the generic error message was not any help.