davidferguson/pibakery #104: Installer can not copy raspbian-pibakery.img

blackhatparas, 2017-06-07:


blackhatparas, 2017-06-07:
hi there , i am new to user and therefore I got myself a raspi
well a month ago I read about your pibakery util so i gave it a try but whenever i run the win installer , it just  shows the error that the raspbian-pibakery.img and raspbian-pibakery-lite.omg can not be copied because the file is corrupted.
i have downloaded the official installer from the official website.
plss reply with a solution as fast as possible 

My deepest apologies for grammatical mistakes and an gratitude for issues section.
Thank You

davidferguson, 2017-06-25:
Hi,
Sorry for the delay in getting back to you. I've tried installing PiBakery on multiple Windows computers, and haven't come across this issue, leading me to think that maybe the installer you downloaded has been corrupted? Could you try downloading the installer and running it again and report back.