davidferguson/pibakery #105: Can't open the write interface on Windows 10

rj93, 2017-06-19:
When ever I click on the write button it doesn't work.

When the program opens I get the following messages every time:
```
Error: ENOENT: no such file or directory, open 'D:\PiBakery\blocks.xml'(…)
Error: ENOENT: no such file or directory, open 'E:\PiBakery\blocks.xml'(…)
Error: ENOENT: no such file or directory, open 'F:\PiBakery\blocks.xml'(…)
```

And each time I click on write I get the following error:
`Uncaught TypeError: Cannot read property 'path' of undefined`

davidferguson, 2017-06-25:
I've had a look at this (had to borrow a Windows 10 laptop, as I use Windows 8) and I think I've found and fixed the issue. Could you download and install [the latest version (0.3.6)](https://github.com/davidferguson/pibakery/releases/download/v0.3.6/PiBakery.Setup.exe) and tell me if that fixes the issue?

rj93, 2017-06-29:
@davidferguson I just installed the latest version, and you have indeed fixed it

closed 2017-06-29 by davidferguson