davidferguson/pibakery #108: Allow copy paste for block arguments

mrosseel, 2017-07-02:
Maybe it's a local problem, but I can't copy paste passwords, network names, file names into the parameter fields of the blocks.

Is this normal? If so, consider this a feature request to fix this, because it's tedious to type long passwords/filenames by hand.

exocode, 2018-07-28:
What I found out: It's possible to drag-and-drop it into the field. (I use a mac)

davidferguson, 2018-08-31:
PiBakery v2 has a different copy-paste mechanism, which should now be working. I've not created a release for this yet, as I hope to fix a few more things first.

theronic, 2019-04-21:
This is especially troublesome for SSH keys.

tvories, 2019-11-18:
Same issue on mac.  Any chance for an update to the v2 release?

RobertStickney, 2020-09-15:
Paste is still not working in v2 on Mac Catalina 10.15.6.  Drag & drop working.

raiman264, 2020-10-26:
just found that it is already fixed but the release binaries are not updated, you need to install from source