davidferguson/pibakery #112: Raspbian Stretch

AdnanHodzic, 2017-08-26:
[Raspbian Stretch has been released](https://www.raspberrypi.org/blog/raspbian-stretch/?utm_source=dlvr.it&utm_medium=twitter), yet when we write images with PiBakery version of Raspbian will be Jessie.

Could you please update PiBakery to provide Stretch images?

Thanks!

PeterChauYEG, 2017-12-28:
Hi,

I was able to get Raspbian Stretch by adding the image to the os folder as specified in the 'build from source' [instructions](https://github.com/davidferguson/pibakery/).

Here's my config in images.json:
```
[
	{
		"compressedFilename": "2017-11-29-raspbian-stretch.zip",
		"compressedMD5": "e7b86f2f0e8654ae47d0f051cdb5bdfe",
		"displayName": "Raspbian Full - Stretch",
		"downloadUrl": "https://downloads.raspberrypi.org/raspbian_latest",
		"filename": "2017-11-29-raspbian-stretch.img",
		"installed": false,
		"skipVersion": 0,
		"uncompressedFilename": "2017-11-29-raspbian-stretch.img",
		"uncompressedMD5": "8fdf6c6c4a3aaa263cfc3912d43a46e2",
		"uncompressedSize": 4919918592,
		"version": 4.9
	}
]
```

What OS are you on? I haven't tried but when I open up the MacOS binary, I see that I can add additional OS' there. I'd assume the same is true for windows.

https://github.com/davidferguson/pibakery-raspbian/issues/46

note to future me: get MD5s with `openssl md5 <file>`.

davidferguson, 2017-12-28:
Current status: see davidferguson/pibakery-raspbian/issues/46

davidferguson, 2018-01-02:
Update for existing installations has been pushed out and new downloads of the installers now contain Stretch. Apologies again for the delay.

closed 2018-01-02 by davidferguson

pgerhardt, 2018-01-04:
Currently, you still need to modify the Raspbian images to read from /boot/PiBakery/ through a systemd service before releasing them because there is no way to do this on the stock Raspbian build?

If we could do this, one could reduce the PiBakery download size significantly, correct?

Rojohnson, 2018-02-20:
Hey @davidferguson I know you have converted everything to stretch but is there a download for Jessie or a way to prevent it updating to stretch? I'm trying to install software that doesn't work with Stretch. Thanks

davidferguson, 2018-02-20:
@Rojohnson Yes, that should be possible. What you'll need to do is download your desired Raspbian release (the two .7z files) from [the PiBakery-Raspbian repo](https://github.com/davidferguson/pibakery-raspbian/releases) (the latest non-Stretch release is 2017-06-21) and unzip them.

They'll each produce a .img file. Rename these .img files to remove the `new` part in their name (so `raspbian-pibakery-new.img` becomes `raspbian-pibakery.img`) and copy these into the PiBakery `os` directory, replacing the existing files.

The `os` directory will be located in `/Library/Application Support/PiBakery/os/` on Mac, and `C:\Program Files (x86)\PiBakery\resources\app\os\` on Windows.

This *should* make PiBakery use those images instead of the Stretch ones. And if you ever want to go back to Stretch, just repeat the above process but download the latest .7z files from the PiBakery-Raspbian repo instead.

----

And in the next major release of PiBakery, you'll be able to drag in any .img file you want to PiBakery, and it'll write that. It may be a little while before this new release is ready though.