davidferguson/pibakery #113: Cannot write to SD card

usdharris, 2017-10-31:
I'm trying to write an image using Windows 7 and Raspian Lite to a microSD 32G card. I'm wanting to set up a boot to a Pi Zero setting the wireless SSID and PW for a headless boot. Unfortunately, when I attempt to write the image I get this error in the inspector. I have tried both Lite and Full versions and I have reinstalled the PiBakery twice. Hopefully I can get this straightened out. Thanks.

C:\Program Files (x86)\PiBakery\resources\app\node_modules\bluebird\js\release\async.js:61 Uncaught Error: Command failed: wmic logicaldisk get caption,volumename,drivetype
'wmic' is not recognized as an internal or external command,
operable program or batch file.

davidferguson, 2017-10-31:
Sorry to hear you're having issues with PiBakery.

The reason for this is probably that the ``%SystemRoot%\System32\wbem`` directory isn't in your PATH environment variable. That's interesting to note as when writing PiBakery I'd made the assumption that it always would be.

There are two fixes:
1. In the next version of PiBakery I'll include an update that means that it will work without the assumption that ``%SystemRoot%\System32\wbem`` is in the PATH variable

2. In the meantime, the easiest fix is to simply add that string to the path variable - see [this SuperUser answer](https://superuser.com/a/1178758/346507) on how to do this but it basically comes down to:
    1. Open **My Computer**, right-click that window and click **Properties**
    2. In the System Properties window, click on the **Advanced** tab.
    3. In the Advanced section, click the **Environment Variables** button.
    4. In the Environment Variables window, highlight the **PATH** variable in the **Systems Variable** section and click the **Edit** button.
    5. If there's not already a semicolon at the end of the PATH, then append one at the end
    6. Then append ``%SystemRoot%\System32\wbem`` at the end

closed 2017-10-31 by davidferguson

usdharris, 2017-10-31:
Thank you. That did the trick. I was at least able to save the image. Unfortunately the image failed to boot the pi zero though. I worked around and loaded a stretch lite version and created a wpa_supplicant.conf file to make the connection. I’ll try to get back to it and try the PiBakery version later and see where I must have messed up. Thanks again though.